jgyates/genmon #792: mqtt plugin stops working after first message (sort of)

MiddleSiggy, 2022-10-17:
I am having issues where the MQTT plug-in is not working after the first send of data.

Let me explain:

I can restart my pi, and it will connect and show everything correctly in HA, but then after a few minutes everything goes to unavailable. (This is because I think things time out, but after a deeper dive, it looks like only the first message makes it through)

Now this is not to say it is not working 100% because I am still getting some messages. I am using MQTT Explorer and I can see the following messages come through:
- generator -> status -> time -> monitor time
- generator -> monitor -> Generator Monitor Status -> Run Time
- generator -> monitor -> Communicator Stats -> Packet Count
- generator -> monitor -> Platform Stats -> CPU Temperature
- generator -> monitor -> Platform Stats -> CPU Utilization
- generator -> monitor -> Platform Stats -> System Uptime
- generator -> monitor -> Platform Stats -> System Time

All other messages never show up after the first transmission..

Also if I change a setting, say set the polling from 2 sec, to 3, and save, it will send 1 successful message, then stop, and only send the list of events above.

I tried to look for a debug log or something, but cannot find anything.

For reference I have 39 devices successfully sending data to the MQTT Broker in HA and they are all working as expected.

Let me know how I can help in debugging this. Or just re-install..?

Thanks

jgyates, 2022-10-17:
With genmon and add-ons the first place to look would be the log file: /var/log/genmqtt.log

Your issue sounds like your MQTT server and the genmon MQTT client are not configured to match. With MQTT you can set a timeout for each value (when does it expire). The default for the genmon mqtt plugin is that values do not expire. See the add on value "Flush Interval" on the settings page under the MQTT plugin. If you set this value to 10, it will send all the values every 10 second regardless if they have changed or not. If you set it to zero it will send all the values when it starts and then only values that change.

To reduce your network traffic you can configure your MQTT server to retain the values and keep the "Flush Interval" to zero.

jgyates, 2022-10-17:
I am going to close this issue but feel free to post to this thread with questions or status.

closed 2022-10-17 by jgyates

MiddleSiggy, 2022-10-17:
It was the "Flush Interval" just like you sad, it would send the first entry, and then time out after 300 seconds because I had it set to "0" zero.

Thank you it is working now.