jgyates/genmon #793: CPU temp on non-Raspberry Pi not displaying

skipfire, 2022-10-17:
I have a NanoPi that I've put in and I have GenMon working on it, but the temp is stuck at 0.  Looking at the code in myplatform.py there is a chunk of code that causes it to exit.
```
if not self.IsPlatformRaspberryPi():
    return DefaultReturn
```
However, further down that file in the exception handler is a comment stating:
`# for non rasbpian based systems` 
which makes it look like it was intended to work with non-Raspberry Pis.  Locally I removed the if that caused it to exit and it displays the temp correctly.  Both paths from the exception handler work.

I did try dropping the if locally and it worked, not sure what impact that may have in other platforms though.

jgyates, 2022-10-17:
From the console, if you type:

     cat /proc/cpuinfo

Show the output of this command. I can potentially add the needed info the IsPlatformRaspberryPi function so the NanoPi is recognized as a Pi type device. If there are other things that need to be documented about the NanoPi with genmon let me know and I can add it to the wiki.

jgyates, 2022-10-17:
The code is looking at the line that looks like this:

Hardware	: BCM2835

The BCM2835 is the System on a Chip so for genmon to think it is a pi we just need to add the part number to the code.
Currently it looks for one of these: 

"BCM2708","BCM2709", "BCM2835","BCM2836","BCM2711"

jgyates, 2022-10-18:
I checked in an update that should allow the CPU temp on linux based systems. Let me know if it works OK for you. I am going to close this issue for now but let me know how it goes.

closed 2022-10-18 by jgyates

skipfire, 2022-10-18:
That does work, thanks.