jgyates/genmon #794: Communication Failure using Wireless Serial Bridge

alexyao2015, 2022-10-18:
I'm just getting setup for the first time using an ESP32 as a serial bridge. It does appear that things are connected correctly, but I'm not seeing any data.

In my logs I see:

```
2022-10-18 16:34:23,601 : WARNING: Initialization not complete after 30 seconds, possible communication failure. Check cabling.
```

as well as these looping over and over:

```
2022-10-18 16:46:03,399 : Discarding byte slave: 64 : Response Address 
2022-10-18 16:46:03,399 : Error Receiving slave packet for register 05ee
2022-10-18 16:46:04,861 : Discarding byte slave: 5b : Response Address 
2022-10-18 16:46:04,861 : Error Receiving slave packet for register 05f2
```

I'm guessing that it is wired up correctly because I am seeing messages being transmitted and data being received. Just not sure if those are the correct messages. From uart, I see these logs coming through:

```
16:47:28 | [D] | [uart_debug:114] | >>> 9D 03 02 0A 00 01 B8 2C
16:47:28 | [D] | [uart_debug:114] | <<< FF
16:47:28 | [D] | [uart_debug:114] | <<< FF
16:47:28 | [D] | [uart_debug:114] | <<< FB 49 DB 92 49 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
16:47:29 | [D] | [uart_debug:114] | >>> 9D 03 02 0B 00 01 E9 EC
16:47:29 | [D] | [uart_debug:114] | <<< FF
16:47:29 | [D] | [uart_debug:114] | <<< DB 92 92 92 92 D2 0A 08 24 82 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00
16:47:30 | [D] | [uart_debug:114] | >>> 9D 03 02 0C 00 01 58 2D
16:47:30 | [D] | [uart_debug:114] | <<< FC
16:47:30 | [D] | [uart_debug:114] | <<< DB 92 92 92 93 00 00 00 00 00 80 04 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
16:47:31 | [D] | [uart_debug:114] | >>> 9D 03 02 0E 00 01 F9 ED
16:47:32 | [D] | [uart_debug:114] | <<< EB DB 93 92 D2 D9 03 00 06 00 80 04 00 00 00 00 00 00 00 00 00 00 00 00 00
```

[image]

Any help would be highly appreciated.

### Your Environment

- Generator Model: Honeywell 7065
- Genmon Logs: [genmon_logs.tar.gz](https://github.com/jgyates/genmon/files/9815315/genmon_logs.tar.gz)
- Genmon Version: V1.18.14

jgyates, 2022-10-18:
For the data:

Valid modbus packet sent from genmon >>> 9D 03 02 0A 00 01 B8 2C
Not  a valid response <<< FF
Not  a valid response <<< FF
Not  a valid response <<< FB 49 DB 92 49 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Valid modbus packet sent from genmon  >>> 9D 03 02 0B 00 01 E9 EC
Not  a valid response <<< FF
Not  a valid response <<< DB 92 92 92 92 D2 0A 08 24 82 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00

It looks like you are sending data but the response is not being recieved correctly. This could mean a few things

1) you data rate is not correct, It should be 9600, N, 8, 1
2) The ground between the ESP and the controller is not connected

jgyates, 2022-10-18:
FYI, this site is helpful for decoding modbus packets

https://rapidscada.net/modbus/ModbusParser.aspx

for example if you put this:

9D 03 02 0A 00 01 B8 2C

and select Modbus RTU and Request it will interpret the packet

A good simple rule, for Evolution and Nexus (Honeywell is Evolution) the first byte should always be 9D (modbus address)

alexyao2015, 2022-10-18:
Thanks for the help!

I went ahead and checked my grounds and they are all connected. I'm using pin 1 for 5v to the esp and pin 2 for ground. Ground is connected all the way through to the max3232 and esp. 

I also tried a loopback test and was able to get loopback on the esp uart, but nothing is coming back when I try loopback through the max3232. I connected pins 2 and 3 and also tried pins 3 and 4 in case something was reversed. Both didn't get any response. Do you have any ideas here?

jgyates, 2022-10-18:
maybe @gregmac has some thoughts

https://github.com/gregmac/Genmon-ESP32-Serial-Bridge

I have not used the ESP based serial over IP device.

alexyao2015, 2022-10-18:
The esp is working fine from my testing. It appears that either the max3232 is wired incorrectly, is broken, or I'm testing loopback through it incorrectly. Uart which is handled by the esp works fine.

alexyao2015, 2022-10-19:
Got it working! Had to play around a bit with the rx/tx lines. Apparently I needed to connect the max3232 Rx to the Rx on the esp rather than tx.

closed 2022-10-19 by alexyao2015

alexyao2015, 2022-10-19:
@jgyates Do you know of a good place to get a wire outside of the generator? Basically, where is an easy place with a hole that I could slip the antenna through?

jgyates, 2022-10-19:
I don't know if this applies to all models, but mine had a round punch out tab used for mobile link installation.

Here is a video from another user, you can see where they put it. Also, depending on your model, you may be able to use the mobile link instructions as a recommendation on how to get the wire outside.

alexyao2015, 2022-10-22:
Thank you for the help. I played around and found I could slip a wire through where the gas line goes in.

I now have it all working, however I'm getting around 25% timeout errors. I noticed that often times a message gets sent and no response is received. I played around a bit outside the generator using loopback and was able to capture 100% of messages looping back at a high frequency using a script I made. It seems that the issue only occurs in the generator itself where I'm not sure if there's some sort of interference causing an issue. I don't see any extraneous messages, however I see messages getting sent out and ignored in some instances. 

Somethings if the box is just right, I get a high amount of data flowing through from genmon, but it stops if the generator is turned on (which shakes the entire box). I would think it's a cable issue but I was able to get the loopback working with no issues while I was physically shaking the entire setup.

Do you have any recommendations or ideas of what could be causing this?

jgyates, 2022-10-22:
Probably just wifi packet loss. I have added an entry to the wiki on how to mitigate this with three settings changes:

https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP#troubleshooting

Change these settings and let me know what your results are.

alexyao2015, 2022-10-22:
Thanks. I've changed those settings and will report back. 

I am worried that this may not fix it however, since it doesn't seem to be a timeout issue but rather packets getting lost some how. I see from the logs here that there is back and forth traffic but on the occasion there's a sent packet without any response at all, not just a delayed response as mentioned in the wiki. This leads me to believe something is getting lost over the wire perhaps. Does there typically need to be some sort of shielding or otherwise between the data transmission wires?

```
21:21:30	[D]	[uart_debug:114]	>>> 9D 03 02 13 00 01 69 EB
21:21:30	[D]	[uart_debug:114]	<<< 9D 03 02 00 33 28 4D
21:21:30	[D]	[uart_debug:114]	>>> 9D 03 00 4C 00 01 58 41
21:21:30	[D]	[uart_debug:114]	<<< 9D 03 02 00 00 68 58
21:21:30	[D]	[uart_debug:114]	>>> 9D 03 02 35 00 01 88 20
21:21:30	[D]	[uart_debug:114]	<<< 9D 03 02 00 00 68 58
21:21:30	[D]	[uart_debug:114]	>>> 9D 03 02 36 00 01 78 20
<--- Break here --->
21:21:35	[D]	[uart_debug:114]	>>> 9D 03 00 53 00 01 69 87
21:21:35	[D]	[uart_debug:114]	<<< 9D 03 02 00 00 68 58
21:21:35	[D]	[uart_debug:114]	>>> 9D 03 00 52 00 01 38 47
21:21:35	[D]	[uart_debug:114]	<<< 9D 03 02 00 00 68 58
21:21:36	[D]	[uart_debug:114]	>>> 9D 03 00 09 00 01 49 94
21:22:03	[D]	[uart_debug:114]	<<< 9D 03 02 01 8F 28 6C
```

jgyates, 2022-10-22:
does genmon show timeout errors (or any errors on the Monitor page of the web UI?

If you have an evo2 then the packet exchange may not be one tx for one rx in all situations.

alexyao2015, 2022-11-02:
It was showing errors that even hit around 20%. They are now around 0.08% which is great. I found that the first max3232 I was using was defective which was causing the high number of errors.

liltux, 2023-01-04:
@alexyao2015, may I ask how you programmed your ESP32 modbus bridge?  Curious if you are using a repository or if you coded yourself? I am looking for direction in this as a cost alternative.  Thank you for your time.

alexyao2015, 2023-01-04:
I'm using esphome. Here's the config files I'm using https://github.com/alexyao2015/ESPHomeYAML/tree/main/yaml/sensors/generator-monitor

I'm also using a home assistant addon that I created to run it here https://github.com/alexyao2015/genmon-addon