jgyates/genmon #795: Canbus error...  (Likely NOT a GENMON issue - just looking for more info...)

DigIntelligence, 2022-10-19:
GENMON began posted a CANBUS error back in August and I haven't got any logging updates via email since then (exercising messages).  While I do get startup email from GenMon, its not pulling any log data from the generator....  (or maybe no log data is being posted in the generator?).  It does appear to show real-time info regarding utility and generator status just no logging info...  This kinda leads me to believe the issue is probably NOT between the Pi and the bus itself (or involve GENMON).

While I don't think this has anything to do with GENMON (as it has been working VERY well through multiple revisions of the software), I am curious if there is a particular place folks might have seen this before.

I will check for more messages on the generator display, the data connections between the Pi and the generator console, etc. 
 However, before I start going down that road, I'm curious if there is a "typical" place where this type of an issue might manifest itself. 

Anyone have any similar insight here?  ...or is this just something that needs to be "cleared" on the generator console?

Here is a snippet of my current logs just after I rebooted GenMon this morning.  Note that all logging stopped on Aug 17.  This did not coincide with any firmware update etc (although I did go from 1.8.14 -> 1.8.15 on Aug 21 without any issue)

Logs : 

    Alarm Log : 
        08/17/22 14:00:01 Canbus Error 
        06/23/22 08:09:41 Inspect Battery 
        06/22/22 14:12:17 Inspect Air Filter 
        06/15/22 14:12:17 Inspect Air Filter 
        06/08/22 14:12:17 Inspect Air Filter 
        ........

    Run Log : 
        08/10/22 14:12:17 Stopped - Auto 
        08/10/22 14:00:07 Running - Exercise 
        08/03/22 14:12:17 Stopped - Auto 
        08/03/22 14:00:07 Running - Exercise 
        07/27/22 14:12:17 Stopped - Auto 
        07/27/22 14:00:07 Running - Exercise 
        07/21/22 17:25:40 Stopped - Auto 
        ........

jgyates, 2022-10-19:
@DigIntelligence 

I make sure I understand your problem, you are saying that your email messages from genmon stop working for weekly exercise, etc, correct?

One think you can do is to physically change the switch (button) on the generator to the off condition and see if the web interface reflects this change and do you get an email if it does change. then try the same thing with manually running the generator, does the web interface show this and do you get an email.

Both of these test will change the engine status and switch status registers. If internally the controller is not changing the state then you could try power cycling your controller.

Of course we will want to rule out any software / email issues so submitting your logs to me via the About page could not hurt.

DigIntelligence, 2022-10-19:
(Thanks for getting back to me)

No, I'm getting email messages from GenMon as I start it up.  So email appears to be  working.  The email related concern revolves around the fact that I am no longer getting notified of any generator events.  My wife asked me yesterday if the Generator was still running its exercises and I have no way to know.  I don't see anything in the GenMon log, and having been getting email updates, since Aug 17th...

GenMon is not reporting any log info (on the web screen) since a Canbus error was posted on Aug 17th.  Again, I'm guessing this has more to do with the generator itself than GenMon.  I'm simply trying to query the community to see if anyone has seen similar behavior before...  (and what the solution might be)

GenMon seems to be accessing the "live" generator info just fine.  Its the logging info which seems to be missing (since the Canbus Error on Aug 17th).  I expect GenMon is likely watching the log to decide when it needs to send me email but it never see's anything worthwhile as it never sees any new log info come across...

Not sure if a controller reset will clear this, or if its simply an issue with GenMon parsing the log.  Though I would ask the community for any insight they may have...

Thanks,
Ed

>>> jgyates ***@***.***> 10/19/2022 10:25 AM >>>
@DigIntelligence 

I make sure I understand your problem, you are saying that your email messages from genmon stop working for weekly exercise, etc, correct?

One think you can do is to physically change the switch (button) on the generator to the off condition and see if the web interface reflects this change and do you get an email if it does change. then try the same thing with manually running the generator, does the web interface show this and do you get an email.

Both of these test will change the engine status and switch status registers. If internally the controller is not changing the state then you could try power cycling your controller.

Of course we will want to rule out any software / email issues so submitting your logs to me via the About page could not hurt.

jgyates, 2022-10-19:
Here is info on power cycling the controller if you decided that you want to do that:

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

Correct this if this is not the case:
* you get email when genmon starts
* you don't get email or any notifications when exercise occurs (no longer getting notified of any generator events)
* genmon is not displaying the run or alarm log on the Log page in the web interface
* Utility voltage and other controls function as expected

The test above would give data points that this is a genmon or a controller issue. It is possible that the engine status register and switch status register are not updating in the controller. If this is the case then no notifications would be sent except the ones on startup. The test above would validate this.

and send the software error logs to me by clicking the "Submit Logs" button on the About page on the web interface.

I don't have any direct experience with a CANBUS error or the symptoms you are describing. The only recommendations I can make are the one I mentioned above.

DigIntelligence, 2022-10-19:
I just ran home to "clear" the "CANBUS" error on the controller screen.  It was sitting at what appears to be an initial configuration screen.  I completed the "initial configuration" on the generator (clock time set, exercise time set) and the generator tried to start.  When it did this, the controller cleared itself again and left me back at the "initial configuration" screen again...

My guess is that I have a bad battery in the generator (and possible a bad charging circuit) and when it tries to start, the batt voltage drops so low as to reset the controller.  My GenMon Pi is powered via a wall wart to prevent drawing down on the generator battery, so this would explain why the Pi was working even if the Generator battery was marginal.

I think its a bad battery in the generator...  I'm going to call from service and then I'll update the "issue" on Github for everyone's benefit...,

Thanks!
Ed

>>> jgyates ***@***.***> 10/19/2022 1:10 PM >>>

Here is info on power cycling the controller if you decided that you want to do that:

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller 

Correct this if this is not the case:
* you get email when genmon starts
* you don't get email or any notifications when exercise occurs (no longer getting notified of any generator events)
* genmon is not displaying the run or alarm log on the Log page in the web interface
* Utility voltage and other controls function as expected

The test above would give data points that this is a genmon or a controller issue. It is possible that the engine status register and switch status register are not updating in the controller. If this is the case then no notifications would be sent except the ones on startup. The test above would validate this.

and send the software error logs to me by clicking the "Submit Logs" button on the About page on the web interface.

I don't have any direct experience with a CANBUS error or the symptoms you are describing. The only recommendations I can make are the one I mentioned above.

jgyates, 2022-10-19:
ok. I am going to close this issue but feel free to post here with any follow up.

closed 2022-10-19 by jgyates

DigIntelligence, 2022-10-19:
Looks like this may be a bad battery (or charging circuit) at the generator itself.

In checking the controller screen, it was sitting at what appears to be an initial configuration screen.  I completed the "initial configuration" on the generator (clock time set, exercise time set) and the generator tried to start.  When it did this, the controller cleared itself again and left me back at the "initial configuration" screen again...

My guess is that I have a bad battery in the generator (or a bad charging circuit) and when it tries to start, the battery voltage drops so low as to reset the controller.  My GenMon Pi is powered via a wall wart to prevent drawing down on the generator battery, so this would explain why the Pi was working even if the Generator battery was marginal.

jgyates, 2022-10-19:
ok. thanks for the follow up info