jgyates/genmon #797: Tank Utility Monitor

Barrygordon, 2022-10-19:
I understand that Generac has bought out the Tank Utility Monitor.  Knowing their propensity to have things work in house onl, I am wondering if Genmon will work with the Generac Tank Utility Monitor or will it only work with the Generac Cloud?  Generac charges a $50 annual subscription fee.  I am totally against what they are doing.

jgyates, 2022-10-19:
as long as you buy your monitor from https://tankutility.com/ you should be good, however I do not have a data point on a Generac branded monitor.

closed 2022-10-19 by jgyates

jgyates, 2022-10-19:
I have a monitor from tank utility and it still works with the gnemon add on and the tankutility mobile app however I purchased my before generac got involved. 

My generator is diesel but use tank monitor on a propane tank with my HVAC system.