jgyates/genmon #799: I'm an idiot.

jltnol, 2022-10-19:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jltnol, 2022-10-19:
So I freely admit I'm an idiot.  I've read countless pages on this program, have looked at few vids, but for the life of me can't find where or how to install this package on a Raspberry Pi. I'm sure this one single step is easy enough, but I don't see any instructions on how.   I've downloaded the .zip file and extracted it, in hopes of finding an installer, but no luck... and I've even tried the CLI, but that doesn't work either, as it it clearly not designed to do what I'm expecting. 

I'm guessing there is a simply command somewhere that I can use with the Terminal to install this... but I haven't seen it.  I'm SURE it is here SOMEWHERE.... but I have yet to discover it.  If anyone can point me in the right direction, I'd be very grateful.

liltux, 2022-10-19:
sudo apt install git
git clone https://github.com/jgyates/genmon.git
sudo bash ~/genmon/genmonmaint.sh -i

Let it install.

djrallard, 2022-10-19:
GenMon wiki, section 3.3. "Setup GenMon Software"

https://github.com/jgyates/genmon/wiki

jltnol, 2022-10-19:
Got it installed and working. I keep forgetting how to do this, mainly because I don’t do it very often. The few commands you provided were all I needed for success. Thanks!  No need to keep this issue OPEN.

closed 2022-10-19 by jltnol