jgyates/genmon #804: Repeating Message, "Not Receiving Data"

bariba01, 2022-10-22:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Expect to receive text message when the generator is exercising.  Message is transmitted via Raspberry Pi WiFi via Twilio text phone message, received by local WiFi router. 

### Actual Behavior

Receive text message: "Generator Exercising".  Then receive text message, "System Health: Not receiving data". Then receive text message, "System Health: OK".  The "System Health: Not receiving data" message followed by the "System Health: OK" message are received periodically as long as the generator is exercising.  The final message sent is "Generator Ready".  I can access the genmon device online and read all parameters, so the data lines between the Raspberry Pi and the Generac controller are obviously OK.

### Steps to Reproduce (including precondition)

Happens every time the generator is running, weather in the exercise mode or running by manual start.  Messages continue as long as the generator is running.

### Screenshot or Pictures relating to the problem (if possible)
[image]

### Your Environment

- Generator Model: G0062620, S/N 3001722514 Evolution, Liquid Cooled
- Genmon Logs: 
[genmon_logs.tar.gz](https://github.com/jgyates/genmon/files/9845216/genmon_logs.tar.gz)

- Genmon Version: V1.18.15

skipfire, 2022-10-22:
The first thing that comes to with intermittent not receiving data during generator operation is that the generator is physically shaking a connection in and out while running, either the ground or 1 of the 2 data wires.

What is your hardware for connecting the Pi to the generator?

bariba01, 2022-10-22:
Thanks for reaching out Jon.
I have connected the Pi board to the RS485 translator using the small female push-on connectors that fit on the posted pins on the Pi I/O/.  These are obviously not great connectors, and they have been in service for a number of years so might be worn.  You can tell when you push them on the post that they are not a very tight connection.  I would be interested to know if you have found a better connector.  However, I can reach out to the device on the web by using the IP address and port number and read all pertinent information relating the generator.  This type of interrogation never fails.  I can also stop and start the generator remotely and this works every time.  While the generator is running remotely I can interrogate the Pi to retrieve status (see recent attempt attached) and this also never fails.  This leads me to believe that it is not a communications problem caused by a poor connection.  I am getting texts saying that the system is not receiving data, but at the same time I can reach out and read all the data remotely.  I use Twilio for text communications.
I am in North Dakota where it gets down to -25 F sometimes in the winter, but the box housing the unit is sealed and the die temperature of the Pi processor is always at or above 90F, so I suspect the box is perhaps 10 degrees at worst case. Obviously, the Pi is not certified for industrial temperature range, so we are pushing things temp wise at both ends of the spectrum.  These Pis only last a couple of seasons up here and need to be replaced, and they are getting harder to buy.  Have you switched to a Pi4 yet?
Thanks again for responding to my question.
Barry

skipfire, 2022-10-22:
With that hardware there isn't a lot of choice.  The simplest and cheapest thing to do may be to try to find new jumper wires and replace them all.  An alternative is buying one of the PintSize.Me HATs (I make & sell them), I've been selling them for just over 14 months now, but I'd still say try new jumper wires first if you can.

I believe that GenMon caches data through some level of dropouts, though @jgyates would probably need to confirm that.  If GenMon does remember the last data then you might never see a problem on the dashboard.

As for the Pi 4, I know some users are using them, but they do consume more power (though it still isn't a ton of power) and they generate more heat which can be problematic in hot weather, but you probably don't have tons of high temps in North Dakota like we do in the southern states (FL, LA, TX).  I haven't had any problems with the Pi dying from use, though I can't say the same for SD cards.  I find I get about 18 months out of an outdoor micro-SD card, though I have recently switched a couple of mine to Sandisk Industrial hoping that they may last longer.

As you said, Pis are getting harder to find, but I know a few people (myself included) have been making some attempts at non-Pi devices running GenMon, and others use a network serial adapter to keep the majority of the hardware inside and just have the adapter at the generator.

jgyates, 2022-10-22:
@bariba01 

The "System Health: Not receiving data" errors are caused by intermittent serial communications between the pi and the controller. This error has nothing to do with SMS or internet communications, only serial communications. You should be able to view the serial communication stats from the web interface on the Monitor page under "Communication Stats". If it is an intermittent cable you will likely see some timeout errors. @skipfire's suggestion regarding cabeling is a good suggestion as this would be the first thing I would look at. Cables can become loose over a period of time due to vibrations from the generator.

Depending on how often this occurs your may be able to trouble shoot your serial communications. This page has some info on that:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

Let me know if you have any other questions.

bariba01, 2022-10-23:
Thanks everyone for your feedback.  This is an incredible product.  I have one on an air cooled Honeywell unit in Fargo, a water cooled 20KW at our lake home in Minnesota and a 25KW water cooled unit in FL. I will try the cable swap and check the serial com stats.  I will let you know what I find out.

closed 2022-10-23 by jgyates

bariba01, 2022-10-23:
> @bariba01
>
> Let us know how it goes. I am going to mark this issue as close for now but please post to this thread if you have any follow up questions or status on this topic.

Will do.