jgyates/genmon #806: Enhanced Exercise and mqtt

andrewvanlagen, 2022-10-23:
When using the Enhanced Exercise with the Mqtt enabled the  engine status doesn't change

Genmon/generator/Status/Engine/Engine State
i get **running** and **cooling down** but it doesn't go back to **off/ ready** 

unless i do to a Mqtt Flush Interval enable ever hour. 
thanks

jgyates, 2022-10-23:
Please submit your logs, or please provide more details on your setup. I did not see any logs, generator type, or any description of if the same problem exists in the web interface as you describe with MQTT.

andrewvanlagen, 2022-10-23:
Model : G0072239
    Controller Detected : Evolution 2.0, Air Cooled
    Nominal RPM : 3600
    Rated kW : 14 kW

it show ready in the web interface but not via mqtt

andrewvanlagen, 2022-10-23:
logs also sent

jgyates, 2022-10-23:
also, what are your settings for Flush Interval and Blacklist Filter in the MQTT plug in

andrewvanlagen, 2022-10-23:
[image]

andrewvanlagen, 2022-10-23:
when i send **setremote=starttransfer** or from the dashboard it work it work correctly

jgyates, 2022-10-23:
edit the file /etc/genmon/genmqtt.conf and add or edit the value to reflect this setting:

     debug=True

then restart. 

This will log every publish attempt and server confirmation of each entry. This will show if the updated engine state is sent and if it is received by your MQTT server. The results are in /var/log/genmqtt.log and are also included when you send your logs from the about page.

Make the settings change, restart then try to duplicate the issue by setting a custom exercise time. We should be able to match the timestamp in the log entry to the time of the exercise to see if the MQTT data is sent and received.

Once you are finished it is best to remove the debug=True or set it to False as this creates a lot of writes to your SD card that over a long period of time will reduce the life of your card.

andrewvanlagen, 2022-10-23:
i sent you the logs.
i never tested it for exercise but it work for a remote start and stop

jgyates, 2022-10-23:
The logs you sent have not additional debug information in the genmqtt.log file. Did you make the above setting change to /etc/genmon/genmqtt.log?

andrewvanlagen, 2022-10-23:
it in my second log file

jgyates, 2022-10-23:
I received two sets in the last few minutes (three total today). None of them have the additional debug information. Have you looked at the file /var/log/genmqtt.log? It does not contain this additional info. If you see it in the file you can post it here.

jgyates, 2022-10-25:
I am going to close this issue as I don't have any data to do anything, specifically the debug output mentioned above. If you want to pursue this please respond to this thread and we can pick it up again.

closed 2022-10-25 by jgyates