jgyates/genmon #807: Question - Genmon with PowerPact 7.5 KW

wtnh, 2022-10-23:
Hi - I am looking at Genmon after a nasty experience with Generac. I have been using CDMA MobileLink for some time. This was working well, but I was getting repeated emails from Generac warning me about the demise of CDMA at the end of 2022. Then I got a "special offer" for a 4G retrofit ($100 off) so I ordered a new unit. I installed it and called support to get it activated in place of the old unit. When I finally got through, the support person looked up my subscription and generator model (G0069980) and informed me that the new units were not compatible with the controller. Nowhere in the installation instruction or on the website was this mentioned. Argghh. So it goes back. (Since they already know which generator I owned - why sell me something that is known not to work?)

The status lights on the new unit pretty much confirm that the 4G unit can't communicate with the PowerPact controller. All Lights are OK except the controller status light which blinks.

So it would appear that the PowerPact 7.5 KW controller is different - maybe just a serial protocol not supported by the 4G MobileLink.

My question is this - can Genmon be made to work with the PowerPact 7.5 KW? If so, are there any special configuration settings required?

Any insights are appreciated. This looks like a really neat project and I am familiar with using Raspberry Pi's and python (although a bit rusty).

Thanks!

jgyates, 2022-10-23:
yes, genmon supports the PowerPact. I have logs that were submitted for the power pact from 2018. There is nothing you have to that is unique to the PowerPact so just set it up like the docs describe.

jgyates, 2022-10-23:
https://github.com/jgyates/genmon/issues/190

closed 2022-10-23 by jgyates

wtnh, 2022-10-23:
Thanks so much! I will be looking at building one.