jgyates/genmon #810: GenMon on OrangePi Zero2

rrescot, 2022-11-01:
Has anyone been able to successfully run GenMon on an OrangePi? 

I'm curious specifically about the Zero2 as it has a u.fl port that I could use for an external wifi antenna outside an enclosure. I don't want to accidently end up down a rabbit hole by going the OrangePi route. If anyone has any suggested builds (armbian?) or tips that would be awesome too.

skipfire, 2022-11-01:
You may want to check out Appendix Q.  I provided instructions for getting it working on NanoPi running DietPi and RockPi running Armbian.  If you use one of those two OSes then the instructions will likely be pretty close.  I do see that the OrangePi has UART5 tied to the same pins.  One concern, either the pinout on the website is wrong, or they flipped the header so that accessory boards go off the edge of the HAT rather than over the SBC board.  On all the other Pis, the 3.3v and 5v pins would be on the bottom of the header as pictured rather than the top.  I am currently running the NanoPi Neo Air-LTS as I found that and the NanoPi Neo Plus 2 were available on Amazon at fairly reasonable prices.  On non-Raspberry Pi you lose the GPIO input, output, and blink add-ons in genmon though, in case those are important to you.  I will also warn you, going down a non-Raspberry Pi route is likely to be much more difficult as Raspberry has done a TON of fantastic work to simplify the use of their system.

[image]

jgyates, 2022-11-01:
Here are some issues marked "non-Pi"

https://github.com/jgyates/genmon/issues?q=label%3Anon-Pi+

Here is appendix Q:
https://github.com/jgyates/genmon/wiki/Appendix-Q---Using-Non-Raspberry-Pi-Hardware

Another thing to consider, if you use [serial over IP](https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP) the bar for using non-Pi hardware is very low as the system hosting the genmon software would only require linux, python and an IP stack.

madisdev, 2022-11-01:
I had got an OrangePI off ebay for that specific reason. But
unfortunately it was DOA. I ended up soldering on an external wifi
connector to my PI Zero. Which was pretty damn difficult :|

rrescot, 2022-11-02:
> PI Zero. Which was pretty damn difficult :|

I saw similar tutorials about that and came to the same conclusion. 

Thank you @jgyates @madisdev and @skipfire for the feedback! After reading through the various issues, I'm thinking that a real (raspberry) pi might be worth the time savings.

closed 2022-11-02 by rrescot