jgyates/genmon #814: Installation Script Error

adrianoden, 2022-11-04:
### Expected Behavior

Following the installation script as documented.   install cryptography step...

### Actual Behavior

building wheels for collected packages: cryptography
  **Building wheel for cryptography (PEP 517) ... error**
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/tmprd0byu2p_in_process.py build_wheel /tmp/tmp09xnai0r
       cwd: /tmp/pip-install-_931unhs/cryptography_df1515e753db42bbad41acd55a4bd4e0
  Complete output (191 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-armv7l-cpython-39
  creating build/lib.linux-armv7l-cpython-39/cryptography
  copying src/cryptography/fernet.py -> build/lib.linux-armv7l-cpython-39/cryptography
  copying src/cryptography/exceptions.py -> build/lib.linux-armv7l-cpython-39/cryptogra

### Steps to Reproduce (including precondition)

Fresh Rasberry Pi installation - 32bit - Hardware raspberry Pi 3

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2022-11-04:
Hi @adrianoden 

Unfortunately I have seen this happen before. This is caused by conflicts between some of the 3rd party software that the project uses. These problems eventually get resolved by the respective third party software maintainer. In the mean time, I would recommend these steps to see if they resolve the issue.

1) Make sure you are using the latest version of the Raspberry Pi OS
2) Update your pi software with these commands:

        sudo apt-get update
        sudo apt-get distro-upgrade

then try running the install script again.

3) If these steps do not work, then try the 64-bit version of the operating system. In the past when I have seen this, the 64 bit version always worked.

jgyates, 2022-11-07:
@adrianoden 

Let us know how your install went. I am going to close this issue now but feel free to post any followup questions or status here.

closed 2022-11-07 by jgyates

adrianoden, 2022-11-07:
Thank you both for replying to the issue. I tried the 64 bit OS and ran into the same problem as before. What ultimately worked was Skipfire's recommendation to run and install as root. 

I'm still getting a serial error and I am not sure if this is an artifact of the installation or if I have a physical cabling issue. Any advice on the log entries below?

[myserial.log](https://github.com/jgyates/genmon/files/9952936/myserial.log)

jgyates, 2022-11-07:
are you showing serial devices when you type

    ls -la /dev/serial*

This page shows the desired output:

https://github.com/jgyates/genmon/wiki/3.2--Setup-Your-Pi-and-Serial-Port

this page has general serial troubleshooting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

You logs look like there could be a device issue with the OS. These two errors are concerning:

2022-11-05 17:25:24,300 : Error in SerialDevice:Write : /dev/serial0:write failed: [Errno 9] Bad file descriptor : myserial.py:244
2022-11-05 17:25:24,302 : Error in SerialDevice:Flush : /dev/serial0:argument must be an int, or have a fileno() method. : myserial.py:223

This make me think your serial ports are not enabled.

adrianoden, 2022-11-07:
Output of my Serial below.  Per documentation I figured I had this correct. If not I was going to re-image and start over. 

```
Last login: Fri Nov  4 17:52:13 2022 from 192.168.0.190
admin@genmon:~ $ ls -l /dev/serial*
lrwxrwxrwx 1 root root 7 Nov  4 17:59 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Nov  4 17:59 /dev/serial1 -> ttyS0
```

jgyates, 2022-11-07:
Did you run the setup script (genmonmaint.sh -i) that sets up the serial port? This run the program serialconfig.py in the OtherApps folder:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#otherappsserialconfigpy-optional

jgyates, 2022-11-07:
The serialconfig program removes any conflicts with the console access to the serial port and also removes conflicts with the bluetooth drivers.

adrianoden, 2022-11-07:
Per the output, things look fine 

```
admin@genmon:~/genmon/OtherApps $ sudo python serialconfig.py -c

Checking : Is enable UART in boot config: OK
Checking : Is BT overlay disabled: OK
Checking : Serial console command line removed: OK
Checking : Serial console service disabled: OK
Checking : BT service disabled: OK

Serial port settings are OK. A reboot is needed if changes were made.

admin@genmon:~/genmon/OtherApps $ 

```

jgyates, 2022-11-07:
Then I would disconnect from the generator, unload genmon and run the loopback test mentioned in the serial troubleshooting link above to validate your serial port is working.

adrianoden, 2022-11-07:
The serial test passed, I ran it several times with no issue.  Any other suggestions?  Does this serial log have anything useful? 

```
[genmon_logs (7).tar.gz](https://github.com/jgyates/genmon/files/9953932/genmon_logs.7.tar.gz)
```

adrianoden, 2022-11-07:
[image]

jgyates, 2022-11-07:
Now that I have all of your logs, it looks like the error that was concerning above was before your last attempted load of the software. The last load of the software does not show this error so I am assuming this is a cable or serial ground issue.

Not sure what image you tried to upload but it does not work.

If you validated the external (looped back through your cable), then I would try to check your ground connection in your cable. The serial connection will not work if the controller and pi are not using a common ground.

I am assuming that you are getting zero packets received. Let me know if that is not the case. This info is on the Monitor page on the web interface.

adrianoden, 2022-11-07:
[genmon_logs (8).tar.gz](https://github.com/jgyates/genmon/files/9954081/genmon_logs.8.tar.gz)

I'll upload a video as well, but everything is powered by the battery within the Generator.  Appreciate your time and willingness to help!

adrianoden, 2022-11-07:
https://www.dropbox.com/s/7yylenqm4tkwm8x/IMG_2002.MOV?dl=0

jgyates, 2022-11-07:
https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

adrianoden, 2022-11-07:
Looking at this guide again, I have what I think is the evo-2 controller (this is a brand-new generator unit). Should I be using pin 5/6 and not pin 7/8 on the Molex connector?

adrianoden, 2022-11-07:
Thanks, I tried swapping RX/TX at the convertor. I'll add a ground to pin5 and see if anything improves. Thanks again

jgyates, 2022-11-07:
Also, to upload and image you must use the github web interface. If you try to send the image as an attachment to an email it will be stripped off. A picture of your connections from your green screw terminal to your molex connector would be helpful. A mistake that is make frequently is assuming the wrong pin numbers on the connector. The diagram of the connector on the cable page states "NOTE: The above diagram is of the controller receptacle, not the mating plug". You are connecting wires to the matching plug so you have to transpose the pin numbers to the right orientation.

adrianoden, 2022-11-07:
I tried sending a link to a movie of my setup. Did that help?  I also wired the ground using the same ground pin the wireless module used. Still no luck. 

Image of wireless module. 
[image]

Image of my molex plug
[image]

skipfire, 2022-11-07:
It could just be the pictures, but it looks like you are trying to use the same pins that the WiFi uses.  If so, you are on the wrong half of the connector as the WiFi module uses RS485 and GenMon uses RS232.

jgyates, 2022-11-07:
I agree with @skipfire , the picture angel is from the wrong side of the connector to be certain but it looks like you have the wrong orientation for your molex connector so you have connected to the wrong pins.

This is a picture of the pinout for the controller connector (not the plug you are using)
[image]

This is a picture of the connections you should make:

[image]

If you are looking at the plug on the back (the side the wires go in) and the tab is facing up, then the pinout on the back of the connector would match the diagram above. In this orientation, the correct wiring would have pin 7 and 8 on the top left for Tx and Rx. From your picture, I looks like you have red to pin 1, black to pin 2 and blue to pin 5 or 6 (hard to tell from your picture angle)

adrianoden, 2022-11-07:
Well, I made the suggested change, I probably had the serial cable wrong, however... still no dice. On top of this, I now have a dead battery. Is it recommended to power the RPI3 from the battery itself or from some other source within the generator?   

I am about to lose all hope on this project :(   - Thank you for your continued support.

skipfire, 2022-11-07:
The battery is the primary method of getting power, the Evolution 2 also has limited current 12v output available in the Molex cable and can be used depending on the Pi you use, a 3B+ and 4B both draw too much power for the controller's 12v though.  Past that you are looking at options like PoE, or DIY throwing in a solar panel and battery to power the Pi.

skipfire, 2022-11-07:
Also, if you have the right pins but the 2 data wires backwards you will still not get data so you might flip those 2 wires within the right spot.  Looking at the plug it should be the top (clip up) two pins for data.

jgyates, 2022-11-07:
Here is a pic of a cable using pin 7 and 8

https://raw.githubusercontent.com/jgyates/genmon/master/Diagrams/Cable%20(End%20that%20connecets%20to%20Evolution%20Controller).jpg

What is symptom that is telling you your battery is dead? Is there an alarm on the controller? Some of the battery alarms are warnings in the controller are just timed response (Check Battery) and some are actual errors  (Low Battery) that occur after at least one 18 hour charge cycle did not make the battery voltage go above a predefined voltage once the charger is deactivated. 

Yes the battery is recommended as it provides the most stable source of power. I have been using mine for years (going on 7 years as of 2022). This is also how generac mobile link is powered (1st gen mobile link). The battery is connected to the charger which is connected to your utility voltage. In normal operation the battery goes between 18 hour charge and no-charge cycles. Of course I can't really know what would happen if you had incorrect wiring on the controller for a long period of time. That could drain your battery but it should charge again unless your battery is old or damaged.