jgyates/genmon #815: I need some help!

ripvega, 2022-11-07:
I purchased two of the HAT boards. tow 12V battery leads and two Molex cables from [pintsize.me](https://pintsize.me/).

I had the Genmon webpage running and configured, all good so far.

I then connected the Molex cable and the HAT to my Evolution 2.0 controller and bang! Two tracks and the rectifier were taken out.

Rpi 4 was OK and I repaired the tracks and the rectifier. The Rpi booted up and all good again whilst not connected to the generator.

To mitigate a further catastrophe I connected just pins 7 and 8 on the controller to the HAT, as that should be all that is required for communication between the two.

No communication.

I replaced the repaired HAT with the spare. Same result.

Another generator that I had been waiting for arrived. I replaced the RP1 4 with a Rpi3 and connected just 7 and 8 to the new generator.

Same result. No communication.

Both of the grenerators are new, Pramac branded, Generac 13kw with Evolution2.0 controllers.

I don't know where to go from here.

Possibilities:

    1. I have two, brand new, faulty Evolution 2.0 controllers.
    2. I have two faulty HATs.
    3. I am missing something obvious.

What concerns me is that the first HAT connection caused the track and rectifier to blow. I am loath to try connecting all 8 pins from the Molex to the new generator and HAT, for obvious reasons. The docs suggest that this is not necessary anyway.

Is there a way to test the HAT independently of the generator?

I'd really appreciate some advice on how to move forward with this.

jgyates, 2022-11-07:
This page details how to test your serial connection (while not connected to the generator) and has pointers to validate hardware. This applies to both the standard configuration as well as the pintsize.me HAT

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

skipfire, 2022-11-08:
Just to be clear from your first post up here, when you had the "bang" moment was it just the Molex connected, or Molex and barrel?

The lowest risk test would be setting both red switches on the HAT to ON, plug in only the Molex (not the barrel) and the fan should spin up.  If that works, disconnect the HAT from the Molex, set your generator to off, add the Pi onto the HAT, then reconnect just the Molex, you should see some LEDs on the Pi light up, then disconnect it from the Molex.  The Pi 4 draws enough power that it throws off the voltage calculation in the controller and will make it think it's a power outage, so I would not leave that in place beyond a couple of minutes of testing.

I'm not suggesting you go buy stuff to try to replicate this, but since your initial post was asking about testing the HAT, depending on what equipment you have laying around, you can setup stuff on a bench and test the HAT and Pi outside the generator, which is what I do with testing every single HAT that I ship out.  You do not have to be as fancy, but my test harness is a diode between 7 & 8 to ensure directionality of the serial connection (you could just short the data pins), and then power on 2 & 3, and I test powering through both the Molex and the barrel with a 12v 3A wall adapter.  I have a script that tests the function button, all 3 parts of the tri-color LED, and does a loopback serial test.

ripvega, 2022-11-08:
`Just to be clear from your first post up here, when you had the "bang" moment was it just the Molex connected, or Molex and barrel?`

The barrel was connected to allow the Pi to boot up and to check it was on line. Then the Molex was plugged in. Instant bang!

Both red switches are set to 'ON'

`test harness is a diode between 7 & 8 to ensure directionality of the serial connection`

Which way around?

I have tested it again this morning with a common 0V as well as 7 and 8 connected. Still no communication.

The firmware on both generators is 1.15.012. As both generators are new, is it possible they have a recent firmware that blocks the connection to Genmon?

ripvega, 2022-11-09:
I have it on my desk to test the loopback.
What is the default user and pw for terminal access?

ripvega, 2022-11-09:
Please could someone confirm that this works with Evolution 2.0 controller firmware 1.15.012?

ripvega, 2022-11-09:
Thank you.

Running `python3 serialtest.py`, got me this:

  **Using python serial library V3.5b0
  Serial port opened
  write data: sent test string
  waiting to received data....
  PASSED! Loopback successful**

So, the board seems to be working.

I put a scope on the Evolution controller 7&8 pins, referenced to each other and to GND, and got nothing.

It seems like the controller is not communicating on those lines. 

It would really help if someone could confirm that this does work with Evolution 2.0 controller firmware 1.15.012.

Is there anything that needs to be set on the controller for this to work? Under the dealer menu maybe?

jgyates, 2022-11-09:
I have working logs for many Evo 2.0 with firmware 1.15. The minor version (012) is not reported via the modbus.

No settings on the controller are needed. It is rare, but I have see a controller or two that has a bad RS-232 interface. Since the controller is a slave modbus interface it will not transmit any data on pin 8 until it receives valid modbus data on pin 7.

I doubt this is the issue but can not hurt: I noticed you have a beta version of the serial python library (V3.5b0). I have seen that happen before, not sure why. Try this:

   sudo pip3 uninstall pyserial
   sudo pip3 install pyserial

Then run the loopback test again and see if the version is V3.5 instead of V3.5b0

ripvega, 2022-11-09:
Thank you.

I’ll give that a try. I’m just using the preloaded as cards that came with
the HATs so I don’t know where the beta could have come from.

jgyates, 2022-11-09:
I am not sure, I think the beta can get in there if some of the servers the package is downloaded from are not available, it can fallback to a mirror site that has not been updated.

ripvega, 2022-11-09:
I'm getting this:

***@***.***:~/genmon $ sudo pip3 uninstall pyserial
Found existing installation: pyserial 3.5b0
Not uninstalling pyserial at /usr/lib/python3/dist-packages, outside
environment /usr
Can't uninstall 'pyserial'. No files were found to uninstall.

ripvega, 2022-11-09:
> 7 & 8 to ensure directionality of the serial connection (you could just short the data pins), and then power on 2 & 3

Is there a reason that you specifically use pin 2? Aren't 2 and 4 common?

jgyates, 2022-11-09:
it looks like your pyserial was not installed using pip3, but probably apt. Not sure how that occurred. The genmon install script uses pip3.

Try this:

       sudo apt-get uninstall python3-serial
       sudo pip3 install pyserial

skipfire, 2022-11-09:
> > 7 & 8 to ensure directionality of the serial connection (you could just short the data pins), and then power on 2 & 3
> 
> Is there a reason that you specifically use pin 2? Aren't 2 and 4 common?

Pin 2 is always grounded, pin 4 seems to be not connected internally on some of the older controller models.

ripvega, 2022-11-09:
That just gets me:

E: Invalid operation uninstall

I'm lost as to how this happened as I'm using the preloaded cards and this
is the first time I've accessed them.

jgyates, 2022-11-09:
sorry, try

     sudo apt-get remove python3-serial
     sudo pip3 install pyserial

ripvega, 2022-11-09:
I now get:

oopback testing for serial port /dev/serial0...

Using python serial library V3.5
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

I'll need to physically try it tomorrow. Too dark here now!

Many thanks for all of your help.

I'm sure we'll get there.

ripvega, 2022-11-10:
We have progress!

But it's not good news.

I connected up the undamaged HAT and Pi to the second generator this morning, following the succesful loop back test.

Inserting the supplied Molex cable caused the same blue flash, track destruction and rectifier blowing that led me here in the first place!

And the problem is...

The supplied battery leads are incorrectly made. The connector has the + and - reversed, resulting in the ouside of the barrel jack becoming +ve.

The rectifier on the board takes care of this until the Molex is connected between the controller and the HAT. Then bang!

This has blown the serial communication on both Pis and, I now suspect, the Evolution controllers. All because of a 50cent battery cable.

Here's the fault:

[image]

As you can plainly see the red goes to black and black to red.

Where do I go from here?

I have two damaged Pis ( I did a loop-back test to check), two damaged HATs, and I suspect, two damaged controllers, due to the supplied cable.

Unless it's supposed to be like this. If so, I think I've exhausted all possibilities.

skipfire, 2022-11-10:
Almost every cable I have shipped is exactly the same as what you have and it hasn't caused a problem anywhere else.  The rectifier is there to make the barrel non-polarized and prevent backwards connections from being backwards on the rest of the HAT.

ripvega, 2022-11-10:
Thank you.

So, if I understand this correctly, my money is now being refunded because I have uncovered that what I purchased is not compatible with the Evolution 2 controller?

Please can you show me where the incompatibility is stated.

Is that correct.?

skipfire, 2022-11-10:
To be clear, I'm refunding you because the hardware appears to not be compatible with European Evolution 2 controllers.  I have a North American Evolution 2 controller and it works just fine and a great many others have bought and used the same boards with North American Evolution 2 controllers.  The PintSize site does not make any mention of being compatible with a Pramac branded generator, nor does it offer shipping outside the US & Canada.

ripvega, 2022-11-10:
Thank you. I checked with Pra,ac and there is a difference - the decals!
That's it.

jgyates, 2022-11-11:
I am going to close this issue but feel free to continue to post to this thread with follow up status or questions related to this topic.

closed 2022-11-11 by jgyates

ripvega, 2022-11-12:
I have now followed this:
[image]
What I now see in Genmon is this:
[image]
I have successfully tested the loopback to the end of the cable.
I have reversed TX and RX, just to make sure, with Genmon restarts in between each change.

Are there any other tests I can run before concluding that the controllers on both generators are damaged?

I should add that the controller part numbers exactly match the US Generac part numbers and Pramac have confirmed that the only difference between US and UK generators is the output voltage of the generator, output frequency, and decals.

jgyates, 2022-11-12:
you could try to power cycle the controller.

https://github.com/jgyates/genmon/wiki/Appendix-I-Power-Cycle-the-Evolution-or-Nexus-Controller

I have seen this fix a comms issue issue before.

Also, it is possible that the RS-232 interface on your HAT is bad, but still passes the loopback. Trying another interface with the pi could rule out and issue with both controllers.

Also, I assume that both of these controllers have been registered. Modbus is not fully operational if the setup is not complete.

ripvega, 2022-11-12:
Thank you.

I'm no longer using the HAT, just the TTL/RS232 adaptor and a USB C battery adaptor.

Both controllers are registered, however, I'll reset both, re-register and try again.

jgyates, 2022-11-12:
would not hurt to check the fuse on the front panel too

ripvega, 2022-11-12:
Do you mean the same fuse that is used for the power cycle procedure, or is there another one?

jgyates, 2022-11-12:
yes, the same one for power cycle

skipfire, 2022-11-26:
In case anyone comes across this in the future, the problem ended up being that the premanufactured data cable from the factory had an incorrect pinout.

ripvega, 2022-11-26:
Yes. All's well that ends well.

I can now get down to configuring Home Assistant and MQTT. Expect some questions!