jgyates/genmon #817: Running genmon as a Docker container

JMVS, 2022-11-13:
Hello,

I've seen many folks running genmon within Docker. Could someone please let me have some instructions on how to do so?
My server is running Ubuntu Server x64 20.04.5 LTS. I obviously have Docker and also Portainer. I've been running pre-build images from DockerHub but never tried to create my own.
Any pointers would be appreciated.

Thanks!

jgyates, 2022-11-13:
I don't have any direct experience however this thread contains some links that may be useful from others who have done this:

https://github.com/jgyates/genmon/issues/776

JMVS, 2022-11-15:
I've managed to do it: https://github.com/gregmac/Genmon-ESP32-Serial-Bridge/discussions/9#discussioncomment-4150134
Please note this is currently configured to use Genmon-ESP32-Serial-Bridge. Should work for other configurations but additional setup might be needed.