jgyates/genmon #818: Question - Propane Tank Level Alarm

rwskinner, 2022-11-24:
Question - Is there a notification with adjustable setpoint for low tank level that can be used on the Mopeka or other external sensor?
A deer hit my gas line and cracked it and I lost 200 gallons of propane in 1 day.  I'd like to set an alert in case a freak accident like his happens again that I might be able to save some of it.  That was an expensive lose!

jgyates, 2022-11-25:
No. Currently the notification is fixed at 10 and 20 percent (two separate notices). I can add this to the feature list.

jgyates, 2022-11-25:
https://github.com/jgyates/genmon/blob/master/feature_requests.md

closed 2022-11-25 by jgyates

rwskinner, 2022-11-25:
I sure don't remember seeing an email notification on either and I had 
notifications turned on.

jgyates, 2022-11-25:
The fuel level is checked once every 10 min so if you lost a lot of fuel rapidly then you may not of gotten a message.