jgyates/genmon #819: Genmon CT Hat

andrewvanlagen, 2022-11-26:
What gpio's does your Ct Hat use the mpc3008

skipfire, 2022-11-27:
@andrewvanlagen if you are asking about the PintSize HAT it uses SPI1 with CS0 which ties up IOs 18, 19, 20, 21.

andrewvanlagen, 2022-11-27:
@skipfire @jgyates 
can you please confirm this pinout
MOSI - Gpio 20
MISO - Gpio 19
CE0 - Gpio 18
CLK - Gpio 21

skipfire, 2022-11-27:
That is correct @andrewvanlagen

skipfire, 2022-11-27:
@andrewvanlagen I forgot about the 1-Wire on 4 for temp.

andrewvanlagen, 2022-11-27:
Does the software support 4 or 8 CT

skipfire, 2022-11-27:
GenMon is only utilizing channels 0 & 1 for the CTs.  The HAT has the 8 channel chip, the pHAT has the 4 channel chip.

jgyates, 2022-11-27:
@skipfire  is correct, the software currently only reads two channels. If there was a person who wanted to test I could add a third for a generator that supports three phase.

jgyates, 2022-11-27:
I am going to close this issue but feel free to post here with any follow up questions on this topic.

closed 2022-11-27 by jgyates