jgyates/genmon #820: Monitoring of the propane level in an underground tank

Barrygordon, 2022-11-27:
Your Genmon system is working perfectly.  Most people who have a whole house generator probably have an in ground propane tank.  Mine is a burried 500 gallon tank. Generac sells a monitor for the tank at a ridiculously high price IMHO.  Most buried tanks have what is called an RD3 tank gauge which has the connectivity for a remote tank level monitor (Mine does and the tank is over 20 years old).  i would suspect that in may cases the buried tank is not too far from the genrator (Mine is about 20 feet.   Several companies sell the attachment for the RD3 wch provides wire leads to connect to to measure a varister whhich tracks the tanks propane level.  TI believe the resistance acrtoss the leads varies from 0-90 ohms.  It would be nice If I could somehow connect this to the Genmon RPi and your application could show the fuel lecel in the tank as a percentage. Burying a 3/4 inch sprinler pipe as a conduit between the tank and the genmon should be fairly simple. The tank level sensor wire pair could then be attached to a pair of GPIO pins on the Genmon RPi and your software could monitor the propane level in real time.

I would be interested in such a feature as a future capability of the Genmon system.  Thoughts?

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2022-11-27:
Hi @Barrygordon 

Three people have submitted ways to make your own tank monitor. Genmon supports all three of these, including one that uses the RD3 to directly connect to the PI GPIO. This wiki entry has details:

https://github.com/jgyates/genmon/wiki/Appendix-L-Adding-a-Propane-Fuel-Gauge-to-Genmon

The first two solutions mentioned in the wiki need the Add on named "DIY Fuel Tank Gauge Sensor" to be enabled.

jgyates, 2022-11-28:
I am going to close this issue but if you have any other questions or comments on this topic feel free to post them here.

closed 2022-11-28 by jgyates