jgyates/genmon #822: Support for GPIO binary sensors?

ripvega, 2022-11-30:
I run my generator from 47kgs propane bottles and measuring contents by any means other than weight is innacurate due to temperature changes so I just rely on an auto changeover valve.

It's be great if I could add a sensor to detect when the change over had happened and pass that through GPIO to genmon and then on to Home Assistant via MQTT.

Is this possible now? If not, is support for user configurable binary sensors something that could be added?

I've seen the Genmon GPIO Inputs addon but that seems to be tied to specific commands.

TIA

jgyates, 2022-11-30:
you can add any information you like to the monitor page via an external file named "userdefined.json". If you wrote a simple python program to monitor the GPIO and saved the status in JSON format to this filename then it will show up via MQTT.

The python libraries RPI.GPIO allows access to GPIO and the library json will allow writing a python dictionary to a JSON formatted file.

https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications#displaying-external-information-in-the-web-ui

You could also edit /etc/genmon/genloader.conf so genmon will load and unload the file every time genmon is loaded / unloaded. For example if you file was named mygpio.py and located in the ./genmon/addon folder, then these lines in the /etc/genmon/genloader.conf file would load it:

     [mygpio]
     module = mygpio.py
     enable = True
     hardstop = False
     conffile =
     args =
     priority = 2

Let me know if you have any questions on how to implement.

jgyates, 2022-11-30:
Here is an example of the contents of a userdefined.json file:

```json
{
    "EmonPi": [
        {
            "Total Power": "1828"
        },
        {
            "Power L1": "1316"
        },
        {
            "Power L2": "512"
        },
        {
            "VRMS": "122.54"
        },
        {
            "Pulse Count": "15917292"
        }
    ],
    "Power": {
        "Time": "Wednesday November 30, 2022 07:32:20",
        "Value": "1828",
        "High": "16936",
        "Low": "1204"
    },
    "Temperature Node": {
        "Time": "Wednesday November 30, 2022 07:31:56",
        "Humidity": "52.8",
        "Temperature": "60.8",
        "Rssi": "-66",
        "Battery": "2.8",
        "External Temperature": "6.8",
        "Pulsecount": "1"
    },
    "Solar": {
        "Load": 1810.0,
        "PV": 0.0,
        "Grid": 1810.0,
        "Time": "Wednesday November 30, 2022 07:29:55"
        }
    },
    "Propane Tank": [
        {
            "Tank Name": "Propane Tank"
        },
        {
            "Capacity": 250
        },
        {
            "Percentage": 80.12
        },
        {
            "Temp": 50.355
        },
        {
            "Battery": "OK"
        },
        {
            "Reading Time": "2022-11-28 20:09:07"
        }
    ]
}
```

for the simpliest entry just put:

```json
{
  "External GPIO": {
        "GPIO Value": "Active"
    }
}

I am going to close this issue for now but feel free to post here with follow up questions or comments.
```

ripvega, 2022-11-30:
Thank you for the comprehensive reply.

I'll need to learn a lot to implement that! I'll add it to the list and give it a go though.

Genmon is such a special project.

jgyates, 2022-12-01:
Sorry if it was not clear, but you must create a python program that will sit in a loop and read the GPIO pin you select to use. The python program will rewrite the userdefined.json file every few seconds. The contents of the file would change depending on the state of your GPIO pin. If the GPIO pin is active the python program will write out

     {
        "External GPIO": {
        "GPIO Value": "Active"
        }
     }

If it is not active then it will overwrite the file to be:

     {
        "External GPIO": {
        "GPIO Value": "Inactive"
        }
     }

The GPIO pin you used are defined in you python program. The JSON file only tells genmon what to display. In the above example the monitor page would show a new section named "External GPIO" with a entry underneath of "GPIO Value: Inactive". You can change the names from "External GPIO", "GPIO Value", "Inactive", etc to what ever you like them to be.

Here is what part of my monitor page looks like, not the "External Data" heading, this is the data taken from the userdefined.json file:
[image]

The python program will look similar to the existing add on gengpio.py.

The short answer is that you must write a python script that will re-write the userdefined.json file every 5 seconds or so, updating it to show your GPIO pin status using the words of your choice. Let me know if you have any other questions.

jgyates, 2022-12-01:
I created a python program that should do what you want.

I have attached two files that can be used to accomplish this:

rename the file gencustomgpio.py.txt to gencustomgpio.py
rename the file gencustomgpio.conf.txt to gencustomgpio.conf
Save this file gencustomgpio.py to your pi in the /home/pi/genmon/addon folder
Save this file gencustomgpio.conf to your pi in the /etc/genmon folder
Look on the advanced page and make sure that the "User Defined Data Directory" is set to /home/pi/genmon/
edit your /etc/genmon/genloader.conf file to add these lines to the bottom:

     [gencustomgpio]
      module = gencustomgpio.py
      enable = True
      hardstop = False
      conffile =
      args =
      priority = 2

Then restart genmon.

On the monitor page of the web interface you should see something like this at the bottom:
[image]

You can change the text displayed on lines 159 - 162 in the python file.

The default uses GPIO 27 (pin 13), this can be changed in the /etc/genmon/gencustomgpio.conf file

[gencustomgpio.py.txt](https://github.com/jgyates/genmon/files/10132599/gencustomgpio.py.txt)

[gencustomgpio.conf.txt](https://github.com/jgyates/genmon/files/10132637/gencustomgpio.conf.txt)

Let me know if you have questions.

jgyates, 2022-12-01:
one minor update:
[gencustomgpio.py.txt](https://github.com/jgyates/genmon/files/10132726/gencustomgpio.py.txt)

ripvega, 2022-12-01:
Wow!

Thank you so much for doing that.

I will give it a try tomorrow and let you know how I get on.

FYI, I have a [Clesse auto changeover valve](https://www.bes.co.uk/clesse-two-pack-automatic-changeover-kit-with-opso-21472/) that supports [this Bluetooth monitor](https://www.bes.co.uk/clesse-compacti-remote-cylinder-monitor-25136/) to indicate which bank of bottles is in use. Being Bluetooth, I have to go stand near it to get a signal so may as well just look.

Adapting it to provide a dry contact input to genmon will make a huge difference.

Thanks again for your work.

ripvega, 2022-12-01:
I've managed to upload the py file but get this when uploading the conf file:

```
ashscott@MacBook-Pro ~ % scp /Users/ashscott/Downloads/genmon\ gpio\ input\ files/gencustomgpio.conf genmonpi@10.0.1.233:/etc/genmon
genmonpi@10.0.1.233's password: 
scp: dest open "/etc/genmon/gencustomgpio.conf": Permission denied
scp: failed to upload file /Users/ashscott/Downloads/genmon gpio input files/gencustomgpio.conf to /etc/genmon
```

Is this just a permissons thing or should I be doing something different?

jgyates, 2022-12-01:
transfer the file to your pi in the home directory first, then copy the file with sudo to the /etc/genmon folder. The /etc folder has root permissions so doing this with the genmonpi user will not work.  Once you copy it then you can copy the file to the /etc/genmon folder via the ssh terminal

      sudo cp ./gencustomgpio.conf /etc/genmon

ripvega, 2022-12-01:
It's still the same result:

```
ashscott@MacBook-Pro ~ % scp /Users/ashscott/Downloads/genmon\ gpio\ input\ files/gencustomgpio.conf genmonpi@10.0.1.233:/home/
genmonpi@10.0.1.233's password: 
scp: dest open "/home/gencustomgpio.conf": Permission denied
scp: failed to upload file /Users/ashscott/Downloads/genmon gpio input files/gencustomgpio.conf to /home/
ashscott@MacBook-Pro ~ % scp /Users/ashscott/Downloads/genmon\ gpio\ input\ files/gencustomgpio.conf genmonpi@10.0.1.233:/home/
```

I don't have a password for sudo

Apologies for being a bit slow with this, but terminal stuff has always seemed like a black art to me.

jgyates, 2022-12-01:
can you log into the pi with ssh with this username? The default user is pi, are you using a prebuilt raspberry pi? If so is this the correct username and password. Logging in via ssh will validate that.

ripvega, 2022-12-01:
I'm using a preloaded SD card from pintsize.

I've been using genmonpi and raspberry. pi as username gets rejected.

ripvega, 2022-12-01:
I meant that the pw gets rejected:

`ashscott@MacBook-Pro ~ % ssh pi@10.0.1.233
pi@10.0.1.233's password: 
Permission denied, please try again.
pi@10.0.1.233's password: `

skipfire, 2022-12-01:
genmonpi is the correct username for the images from PintSize.Me.  Aside from the username (changed for security) it is a standard Raspberry Pi OS install with genmon installled per wiki instructions.

jgyates, 2022-12-01:
Type this:

     scp /Users/ashscott/Downloads/genmon\ gpio\ input\ files/gencustomgpio.conf genmonpi@10.0.1.233:/.

and this

     scp /Users/ashscott/Downloads/genmon\ gpio\ input\ files/gencustomgpio.py genmonpi@10.0.1.233:/. 

If you use scp you must either use the full path to your home (/home/genmonpi) or use a relative path (./). Just using /home points to a folder that the genmonpi user does not have permission.

This will put both files in the genmonpi users home folder (/home/genmonpi)

jgyates, 2022-12-01:
sorry, make that:

   scp /Users/ashscott/Downloads/genmon\ gpio\ input\ files/gencustomgpio.conf genmonpi@10.0.1.233:./

and 

   scp /Users/ashscott/Downloads/genmon\ gpio\ input\ files/gencustomgpio.py genmonpi@10.0.1.233:./

the "./" is the relative path of the home folder.

ripvega, 2022-12-03:
Thank you for the update.

The .py file seems to be in the addon folder from my previous attempts:

[image]

When I try the first of the suggestions above, I get:

[image]

The file does not seem to show up in genmon though:

[image]

Running the second one gets me this:

[image]

Though I suspect that this is because I don't have permission to overwrite, for some reason.

The .conf file does show up here though:

[image]

So now I just need to move it to genmon??

ripvega, 2022-12-03:
Do I still need to do this bit?

> You could also edit /etc/genmon/genloader.conf so genmon will load and unload the file every time genmon is loaded / unloaded. For example if you file was named mygpio.py and located in the [...]

jgyates, 2022-12-03:
When you login via ssh you will login to the home directory of the user you are using to log into the pi. I this case your user name is genmonpi. So the home directory for this user is /home/genmonpi. If you are not using the user named pi then you will not reference /home/pi. 

It looks like your copy of gencustomgpio.conf worked and it is now in the location we specified (the home folder of the genmonpi user, /home/genmonpi). You can copy this to the /etc/genmon folder by doing this:

1) log in using ssh with the genmonpi user
2) type this:

        sudo cp ./gencustomgpio.conf /etc/genmon

Your attempt to copy the genmoncustomgpio.py file failed because you put /. at the end instead of ./
In linux the dot slash means current directory. So this specifies the home folder of the user you are using in the scp command. We can skip a step with this file and just copy it straight into the genmon/addon folder with this command:

     scp /Users/ashscott/Downloads/genmon\ gpio\ input\ files/gencustomgpio.py genmonpi@10.0.1.233:./genmon/addon

Once you do this then you must edit the file /etc/genmon/genloader.conf file as instructed above, note this is what you need:

     [gencustomgpio]
     module = gencustomgpio.py
      enable = True
      hardstop = False
      conffile =
      args =
      priority = 2

Do **NOT** use mygpio.py. The module and section (the part in brackets) must match the filename.

Then restart with a reboot or with this command while you are in the genmon folder:

     ./startgenmon.sh restart

The "User Defined Data Directory" should be the location of your genmon files. In your case that is /home/genmonpi/genmon/ so leave that the way it is.

ripvega, 2022-12-03:
Thank you.

Before I edit the file, do both of these now look correct?

[image]

[image]

jgyates, 2022-12-03:
A few things:

You do not have the file gencustomgpio.conf in the right place. It should be in the /etc/genmon folder. You have it in the /home/genmonpi/genmon folder. Not sure how that occurred. Earlier it was in the /home/genmonpi folder (the genmonpi users home folder) and I gave you instructions to copy it to /etc/genmon with this command:

     sudo cp ./gencustomgpio.conf /etc/genmon

You should be able to validate the file is there by typing this:

     sudo ls -la /etc/genmon/gencust*

If you can get that file there then you should restart genmon as instructed above. Once you do restart then you can type this command:

      cat /var/log/gencustomgpio.log

This will print the error log for the program

You should also be able to see if there is a file named "userdefined.json" in the /home/genmonpi/genmon folder.

ripvega, 2022-12-04:
Thank you.

Here's what I have:

[image]

[image]

[image]

jgyates, 2022-12-04:
Almost there.

Now edit the file /etc/genmon/gencustomgpio.conf and change this line:

     output_path=/home/pi/genmon/

to this:

     output_path=/home/genmonpi/genmon/

save the file and restart genmon. The GPIO used is pin 13 (GPIO number 27)

ripvega, 2022-12-04:
Brilliant!

External data now shows up. Thank you so much.

Can I change "Custom GPIO" and "Pin State" in this...

`           MyDict = {"Custom GPIO": {"Pin State": state_message}}`

... to something more meaningful for my application?

If I wanted to use a different pin, what would need to be changed in gencustomgpio.py?

jgyates, 2022-12-04:
Yes, just change the stuff in quotation marks:

```python
          if state:
                state_message = "Active"
            else:
                state_message = "Inactive"
            MyDict = {"Custom GPIO": {"Pin State": state_message}}
```

If you want a different pin then change the entry in /etc/genmon/gencustomgpio.conf

     [gencustomgpio]
     poll_interval=5
     gpio_pin=13
     output_path=/home/pi/genmon/

The entry above "gpio_pin" is the pin number (not the GPIO number). To change the GPIO input then can this to a different pin number.

It will show up in MQTT. The MQTT output matches what is displayed on the web interface so it would show up under genmon/Monitor/External Data/Custom GPIO/Pin State

If you change the names the of the entries in gencustomgpio.py then it will also change the names of the MQTT topic path.

ripvega, 2022-12-04:
Thank you.

I'm guessing that it won't show up in MQTT until there's an initial state change, is that correct?

I don't have the hardware attached to the pin yet (and it's too cold and dark for fiddling outside right now) so can't force a state change.

jgyates, 2022-12-04:
Once the software loads then all data is sent via MQTT (the initial state read counts as a state change).

jgyates, 2022-12-04:
Thanks, I am glad it is working.

ripvega, 2022-12-04:
Well, almost. I've yet to test the hardware and see it show up as an MQTT message, but the hard work is done.

jgyates, 2022-12-04:
Just post any follow up questions to this thread.

jgyates, 2022-12-05:
duplicating everything will not work. The only solution is to modify the py file.

ripvega, 2022-12-05:
I was afraid of that.

I didn't want to force disassemble the changeover valve hardware until I knew that the software side was feasible.

jgyates, 2022-12-05:
I will try to update the program in the next couple of days

ripvega, 2022-12-05:
That will be very much appreciated.

I've been studying gencustomgpio.py and can make sense of some of it but adding extra GPIOs is currently beyond my skill level *as I'm sure you've noticed 😁).

I'll certainly write up the hardware side for others. I'm just awaiting the delivery of some cylindrical reed switches and unipolar hall effect sensors.

jgyates, 2022-12-06:
I have added the gencustomgpio program to the repository and added functionality to make it use any number of the GPIO lines.

To use this you should update your software from the about page. Then log into the pi an execute this command:

    sudo rm /etc/genmon/gencustomtgpio.conf

This will remove the old conf file. The new fill will be copied once you restart gennmon. Once the new file has been copied by the software with a restart you can edit the file /etc/genmon/gencustomgpio.conf

This page will give you details on how to edit the conf file to have the labels you want on each GPIO pin.

https://github.com/jgyates/genmon/wiki/1----Software-Overview#gencustomgpiopy-optional

Let me know if you have any questions.

ripvega, 2022-12-06:
Thank you!

That is outstanding. I'll get to it tomorrow. It opens up so many possibilities.

ripvega, 2022-12-06:
I've not tried the hardware as yet, but so far, so good. Thank you.

[image]

ripvega, 2022-12-07:
I'm now testing out the hardware and not seeing the change in MQTT message.

Using raspi-gpio get, I see:

[image]
 So, GPIO27, pin 13 is seeing the change when pulled high.

My .conf file looks like this:

[image]

The Monitor page shows this:

[image]

And I am seeing the changes:

[image]

But that's not reflected in MQTT:

[image]

Is there something else I need to enable?

jgyates, 2022-12-07:
what are your settings for the MQTT plugin? Also, update to the latest again and then send me your logs from the About page. This way I can look at the logs and see if there is any thing strange. Make sure you update the software first, then try to duplicate the problem, then send the logs. The update is to add the new log file to the list of logs sent when submitting your logs on the About page.

ripvega, 2022-12-07:
I think your reply crossed my post deletion. Sorry!

I've sorted it. Turns out, for this to work ,one has to be logged in to the correct genmon instance. Who knew.

jgyates, 2022-12-07:
ok. let me know if you have any other isssues