jgyates/genmon #823: Outage log MQTT published packet is too long for home assistant sensor

ripvega, 2022-12-07:
I'm trying to extract data from the outage_log that is published via MQTT.

The published string represents the last 8 outages (maybe there would be more but my log only shows 8 so far).

The problem I have is that the published packet exceeds the home assistant sensor character limit of 255 characters so I can't parse to just one event instead of 8 and then extract the pertinent data.

Is there a way around this in genmon?

lmamakos, 2022-12-07:
I've had a similar issue in the past with the maximum length of the value of a "state" in Home Assistant.   As you suggested, if the MQTT payload is in JSON format, you can have Home Assistant parse the JSON, grab one part as the value of the state and have the entire payload turned into attributes.  

Turns out I ran across this myself some years ago before genmon supported MQTT and I used the "rest" platform in Home Assistant to fetch various values.   See https://community.home-assistant.io/t/need-help-storing-url-for-use-in-generic-ip-camera-over-255-chars/99733/2 for an example of doing this sort of thing, if that's helpful.

jgyates, 2022-12-07:
Let me look into this.

ripvega, 2022-12-07:
If only the most recent outage log was sent then it could be parsed using a template sensor, which is how I handle the Alarm Log.

> Turns out I ran across this myself some years ago before genmon supported MQTT and I used the "rest" platform in Home Assistant to fetch various values. See [...]

Thank you for that. I will take a look.

ripvega, 2022-12-08:
This is the outage log payload as valid JSON:

`{"Date": ["2022-12-06 08:52:39"], "Duration": ["3:37:08"], "Estimated Fuel": ["7.07 L"]} `

However, as the payload is currently published with multiple outages it will result in duplicate keys so would need to be published one outage at a time, as per the alarm log.

jgyates, 2022-12-08:
I can make the outage log return a json list of entries:

generator/Outage/Outage Log:["2022-03-16 18:36:26, Duration: 0:00:20, Estimated Fuel: 0.00 gal", "2022-02-25 13:42:09, Duration: 0:33:30, Estimated Fuel: 0.00 gal", "2022-02-24 19:47:11, Duration: 0:01:50, Estimated Fuel: 0.03 gal", "2022-02-08 12:04:05, Duration: 0:00:19, Estimated Fuel: 0.00 gal", "2021-05-10 11:29:16, Duration: 0:00:15, Estimated Fuel: 0.00 gal", "2021-05-10 11:28:22, Duration: 0:00:28, Estimated Fuel: 0.00 gal", "2021-05-10 11:26:26, Duration: 0:00:08, Estimated Fuel: 0.00 gal", "2021-05-10 11:22:32, Duration: 0:00:12, Estimated Fuel: 0.00 gal", "2021-05-10 11:20:15, Duration: 0:01:09, Estimated Fuel: 0.00 gal", "2021-05-10 11:17:26, Duration: 0:00:46, Estimated Fuel: 0.00 gal", "2021-05-10 11:12:20, Duration: 0:00:45, Estimated Fuel: 0.00 gal", "2021-05-10 11:03:59, Duration: 0:00:34, Estimated Fuel: 0.00 gal", "2021-02-10 09:55:29, Duration: 0:00:46, Estimated Fuel: 0.00 cubic feet", "2020-11-12 22:15:22, Duration: 0:00:15", "2018-11-26 06:54:07, Duration: 0:01:36", "2018-11-26 06:53:31, Duration: 0:00:34", "2018-11-26 02:27:06, Duration: 0:00:00", "2018-11-26 00:04:31, Duration: 0:00:34", "-08 16:58:03, Duration: 0:00:00", "2022-08-06 00:43:13, Duration: 0:00:18, Estimated Fuel: 0.00 L"]

If you are OK with the format above I can check it into github.

ripvega, 2022-12-08:
Thank you.

When I run that through the [JSON validator](https://jsonlint.com/), it throws up errors.

I've only tested using one log entry in the HA sensors. If multiple logs are published they would need to be truncated somehow to just the most recent and each piece of data needs a key, like so:

{"Date": ["2022-12-06 08:52:39"], "Duration": ["3:37:08"], "Estimated Fuel": ["7.07 L"]}

{"KEY1": ["DATA1"], "KEY2": ["DATA2"], ..................... "KEYn": ["DATAn"]}

jgyates, 2022-12-08:
I do not understand your last statement. What did you run through the JSON validator? When I run the text above it passes. What specific errors are you seeing?

If truncating the current entry is an absolute requirement I suggest that you blacklist the outage log and skip that entry. Parsing in the format you suggest is not in line with the design goals of genmqtt. The genmqtt add on is designed to avoid and knowledge of the data names as it supports multiple types of generators.

jgyates, 2022-12-08:
are you using the "JSON for Numerics" setting in the genmqtt add on?

jgyates, 2022-12-08:
if you are using the "JSON for Numerics" setting, I could make the mqtt output look like this:

generator/Outage/Outage Log/0/Date:2022-03-16 18:36:26
generator/Outage/Outage Log/0/Duration:0:00:20
generator/Outage/Outage Log/0/Estimated Fuel:0.00 gal

generator/Outage/Outage Log/1/Date:2022-02-25 13:42:09
generator/Outage/Outage Log/1/Duration:0:33:30
generator/Outage/Outage Log/1/Estimated Fuel:0.00 gal

As a log entry was added it would shift every log entry down one number and add the new entry at position 0.

This is the main reason that the maintenance, run and alarm logs are not included in the mqtt output. The "proper" formatting is very subjective. I added the outage data as there was some additional data related to outage on the outage page that is useful.

ripvega, 2022-12-08:
Yes, I do use the JSON numerics setting.

> This is the main reason that the maintenance, run and alarm logs are not included in the MQTT output.

This is where I'm confused then.

I do see the Alarm Log in MQTT output, which is why I thought the outage log could just be the same.
[image]

What have I misunderstood?

> generator/Outage/Outage Log/0/Date:2022-03-16 18:36:26
> generator/Outage/Outage Log/0/Duration:0:00:20
> generator/Outage/Outage Log/0/Estimated Fuel:0.00 gal

This could work. I'd have to give it a try. It's really just the 255 character limit that is making things awkard.

ripvega, 2022-12-08:
> generator/Outage/Outage Log/0/Date:2022-03-16 18:36:26
generator/Outage/Outage Log/0/Duration:0:00:20
generator/Outage/Outage Log/0/Estimated Fuel:0.00 gal

The above contains 153 characters, which is doable.
Would it arrive as three messages or one long message with line break character separation?

I can then give it a test.

jgyates, 2022-12-08:
in the example I used above each log is represented by 3 MQTT messages. One log entry looks like this:

    generator/Outage/Outage Log/0/Date:2022-03-16 18:36:26
    generator/Outage/Outage Log/0/Duration:0:00:20
    generator/Outage/Outage Log/0/Estimated Fuel:0.00 gal

This would have a log entry of generator/Outage/Outage Log/0/ with three parameters Date, Duration, and Estimated Fuel

In theory you could combine these to looks like this:

     generator/Outage/Outage Log/0/"{"Date": ["2022-12-06 08:52:39"], "Duration": ["3:37:08"], "Estimated Fuel": ["7.07 L"]}"

but that requires extra steps to pars the JSON string.

Regarding your question on the logs ("What am I missing?"), the first four menu items in the web interface are Status, Maintenance, Outage, Logs and Monitor:

[image]

The MQTT output is the same as the data displayed in the web UI. Both are converted from the same underlying python data into JSON. That you give above is this MQTT topic generator/Status/Last_Log_Entries/Logs. This is the same data from the Status page which represents **only one** log entry for each of the three logs (the last log entry). The full log (50 entries of each of the three log types for a total of up to 150 entries) are on the Logs page. This data is not converted to MQTT. The outage log can have an unlimited number of entries. The examples you give above are for only one entry. 

     {{"Date": ["2022-12-06 08:52:39"], "Duration": ["3:37:08"], "Estimated Fuel": ["7.07 L"]},
       {"Date": ["2022-12-07 08:52:39"], "Duration": ["1:17:09"], "Estimated Fuel": ["3.02 L"]}, ...}

If you have 50 entries of the above in a JSON string you still have a string that is longer than 255. You have two options:

1) Make all of the log entries in one MQTT topic, but live with a JSON string longer than 255
2) Break up the log into individual log entries but each one must have a unique MQTT topic path (generator/Outage/Outage Log/0/, generator/Outage/Outage Log/1/, generator/Outage/Outage Log/2/, etc). The issue is that the MQTT topic has to be unique for each entry.

jgyates, 2022-12-08:
I have checked in the change so if you have JSON for Numerics enabled then you should see the new format of the outage log. Let me know if you have any issues or questions. I am closing this thread but feel free to post any follow up comments or questions here. Let me know how it goes.

closed 2022-12-08 by jgyates

ripvega, 2022-12-09:
Excellent, that is going to work superbly.

Thank you.

ripvega, 2022-12-09:
That works perfectly.

For anyone else that might need this, the MQTT sensors I defined in HA are below.

[image]

  - unique_id: "7797ec3c-e1c5-4e10-ba9d-f8f15d27a009"
    expire_after: 300
    name: Generator Outage Log Date
    state_topic: "generator/generator/Outage/Outage_Log/0/Date"
    value_template: "{{(value)[:10]}}"

  - unique_id: "7797ec3c-e1c5-4e10-ba9d-f8f15d27a049"
    expire_after: 300
    name: Generator Outage Log Time
    state_topic: "generator/generator/Outage/Outage_Log/0/Date"
    value_template: "{{(value)[11:16] | trim}}"

  - unique_id: "7797ec3c-e1c5-4e10-ba9d-f8f15d27a029"
    expire_after: 300
    name: Generator Outage Log Estimated Fuel
    unit_of_measurement: l
    state_topic: "generator/generator/Outage/Outage_Log/0/Estimated_Fuel"
    value_template: "{{(value)[:4] | trim}}"

  - unique_id: "7797ec3c-e1c5-4e10-ba9d-f8f15d27a039"
    expire_after: 300
    name: Generator Outage Log Duration
    state_topic: "generator/generator/Outage/Outage_Log/0/Duration"