jgyates/genmon #827: Remote location

nickabocker171, 2022-12-09:
I have a generator that is sitting on top of a mountain top. I originally was using the generic mobile link software and found the genmon software. (which I have since deployed to 7 generators and adding three more in the coming months). The generator is the following

Evo 2.0 controller
13 KW model 

Generator has been installed for two years and working great. Since putting the genmon hardware in I have the same issue happen twice.

a week ago it was set to run a scheduled "Enhanced Excerise Quite" at 10:00 am. which set an alarm code for 1902. Researching this we determined a mode change of off to on should resolve the issue. 

Jumping to yesterday I was able to get my team to snowmobile to the generator and. get the unit cycled over. at 11:43 yesterday I had the generator preform a "enhanced excerise normal" (based a research this was recommended for the enviroment. 

Suddenly today the generator ran again at 2:45, im assuming with its built in excerise schedule as the timing matches what genmon reports, and once again locked up and set alarm code 1902. My two questions to you are is there any way to remotely clear this alarm and two why would the generator run both excerise schedules.

nickabocker171, 2022-12-09:
I have just sent over my registers and my logs

jgyates, 2022-12-09:
I have not received any logs yet. Your outbound email must be setup and working for this to work.

nickabocker171, 2022-12-09:
just wanted to let you know first I found out I did not have the email setup on this site.

2ndly the command you described above cleared the alarm I am currently back online!

jgyates, 2022-12-09:
Thanks for the feedback. If you setup email,  can you send your logs? If not can you let me know the firmware version you are running on your Evo2.0. It should be on the Maintenance page. If can possibly add this to the web interface for evo2.0

jgyates, 2022-12-09:
I checked in an update that should add a "Reset Alarm" button at the bottom of the Maintenance page.

If someone with an Evolution 2.0 controller can run these test I can validate they work and add this functionality (if it works) to the web interface. I would recommend that they be physically present at the generator incase you need to manually stop it.

**First update the software to the very latest from the about page.**

Run ClientInterface.py as described above, then run these commands:

This command should set the front panel switch to Off:

      generator: setremote=off

This command should set the front panel command to manual and start the generator:

      generator: setremote=manual

This command should set the front panel switch to Off and stop the generator:

      generator: setremote=off

This command should put the generator in Auto mode:

      generator: setremote=auto

Also, let me know what version of the firmware you are running. The firmware version is on the Maintenance page.

nickabocker171, 2022-12-10:
Let me try to get my email setup, it might be a day or so before I can get you the email as well as test those commands on another generator I have more locally.  The generator that started this however is on firmware v1.15

jgyates, 2022-12-10:
ok. not rush. Thanks

jgyates, 2022-12-10:
I am going to close this issue but let's use this thread to continue the testing discussion.

closed 2022-12-10 by jgyates

nickabocker171, 2022-12-12:
I just sent the logs/registers from another monitor that is identical to the one we started with this ticket. I also updated this one and will try to run the commands above today

nickabocker171, 2022-12-12:
will do thank you! btw does generator: setquite=off not work on air cooled? I tested prior to updating but not since. Wonder if there is a different register for it on air-cooled?

nickabocker171, 2022-12-16:
Hey just wanted to provide some more data is there an email I can send some collected information to?

jgyates, 2022-12-16:
generatormonitor.software@gmail.com

griffths, 2023-01-15:
I have a maintenance alarm active on my controller so I thought I would see if the remote reset commands would work to contribute to the effort.

I have a liquid cooled Nexus based generator and followed the steps in issue #827.  The command appear successful but the alarm is still active from what I can tell without going out and physically looking at the controller display.

[image]

@jgyates would logs help research if this is possible on a Nexus controller?

I am posting here to ask the team if there is anything further that can be tested / attempted before I go outside and clear the alarm manually 

Jay

jgyates, 2023-01-15:
Logs never hurt to send. The main thing to look at is register 0000. If this is zero then the alarm is cleared. If the user interface is still showing an alarm active then the alarm is still active since this info is read from register 0000.

jgyates, 2023-01-15:
another thing to consider is that the maintenance status will still show due even if the alarm is cleared as the software will put change the UI color if maintenance is due, but the alarm status will be cleared and register 0000 should read 00000000.

jgyates, 2023-01-15:
sorry it is actually register 0001 not 0000

griffths, 2023-01-16:
Logs sent

griffths, 2023-01-16:
It appears that registers 0000 and 0001 are still active.

griffths, 2023-01-16:
It appears that registers 0000 and 0001 are still active.

[Genmon register pic](https://user-images.githubusercontent.com/19066716/212577604-f1206779-aa89-4715-9a2e-b3fc96e27084.png)

jgyates, 2023-01-16:
yep, looks like it did not reset the alarm condition. Also, keep in mind that not all alarm conditions are resettable via the remote command, however I would guess that this one (inspect air filter) would be. If you get another type of alarm you could try this again. There is a small possibility it would work on another alarm condition, but a larger possibility that it would not.