jgyates/genmon #828: Generac 17KW Air Cooled BEFORE Nexus

ScanCaster, 2022-12-12:
I am attempting to confirm that this project is viable for MY UNIT. The Generac OEM device is not compatible, so this is pretty much my only option it appears.

Model 55040 17KW Air Cooled Guardian

I have **THIS CONTROLLER: 
[2008 Era Generac Pre-Nexus**](https://raw.githubusercontent.com/jgyates/genmon/master/Diagrams/2008-PreNexusController.jpg) controllers. See [this](https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues) page for more info.

So... is this project compatible.??????? Yes/No?

I've read the appendix... Specifically: Testing has shown that these controllers respond over the serial link in a non-modbus compatible way. To support these controllers you must set "Modbus slave transmit address" on the advanced setting page in the web interface (or response_address in /etc/genmon.conf).

I've also reviewed this: https://github.com/jgyates/genmon/issues/477

So before I start my hunt for the Pi, and purchase a bunch of stuff to go with it like the PiCrust Interface, cable, enclosures, etc... 

I don't want to waste my time or any one else's here trying to get this to work... 

And where is this mystical connector??? As  when I did the recent oil/filter change, battery change etc.... I didn't find it... so am I just blind??? Probably... not looking in the right spot????? Total moron, yes likely..... I just have to be looking in the wrong spot or something....sort of frustrated with this search ....

**_Thank you for the time and input!_**

_PS: I couldn't find the "discussion" tab for this project, so I made an issue...._

lwbeam, 2023-08-20:
@ScanCaster    In case you're still considering this project, I'm certain that Genmon will work with your machine. [I've had it installed](https://github.com/jgyates/genmon/issues/31#issuecomment-747104229) for several years on a Model 0055231 17kW Air Cooled Guardian. To the best of my knowledge your machine is essentially the same as mine, as it appears to have the same engine and controller. Yours just might be a little newer, and came equipped with a different transfer switch. Unfortunately, my machine recently died (oil pump failed), and I've just replaced it with a 14kW Model G0072230. Hopefully you get more mileage out of yours.

karamanos1, 2023-09-05:
@lwbeam Do you still have your controller setup? I have the same Air cooled Guardian as you have (0055231 17kw air cooled)and would like to know which setup you used so I can create step by step. What were you able to do (pros and cons/limitations with the pre nexus)I would even buy yours if you still have it? Any help greatly appreciated. Thanks

karamanos1, 2023-09-05:
@ScanCaster Were you able to gain any traction with the setup for your Model 17kw 2008 pre nexus. I am in similar boat trying to steer the waters on what works or might work and list of materials/needed for connector. Thanks

lwbeam, 2023-09-08:
@karamanos1 If by _controller setup_ you meant my Genmon associated hardware, I've moved it to my new generator. I didn't need to make any modifications to it to get it working, but I did decide to use a new SD card and a fresh install of the operating system and software. I've posted a description of my setup [here](https://github.com/jgyates/genmon/issues/31#issuecomment-747104229), with a complete parts list. Below my initial post there's a follow-up with more details. Have a look at them, and I'll try to answer any specific questions you might have. As for using Genmon with a pre-Nexus controller, it works _right-out-of-the-box_ with one very minor, well documented setting change. On the **Advanced Settings** page you need to set **Modbus slave transmit address** to **41**. Because your machine doesn't have the required sensors, you may also want to set **Disable Power / Current Display** on the **General Settings** page. Finally, it's my understanding that the **Enhanced Exercise** Add-On doesn't work with a pre-Nexus controller. There may be others, but these are the ones of which I am aware.

karamanos1, 2023-09-09:
Great thanks so much for the help. Any monitoring is better than none (the
way i see it even if does minimal) I see you ran ethernet direct. Did you
try it with wifi?

Thanks

Nick

lwbeam, 2023-09-10:
The year before I installed my monitor I arrived at my cottage in late April to find that the generator had failed to start during an exercise run in early January, and had been unavailable for nearly four months. Thankfully it wasn't required during that time, even though it had been a brutal winter. The person I had checking my place weekly was only looking to see if the light was on on the end of the generator, and didn't understand that the actual colour was important. The next year he got a much more detailed pre-job briefing, that used the _three-way communications_ protocol. Yes, some is **way better** than none.

No, I didn't bother trying Wi-Fi. My network router is over 100 feet from the generator, and I assumed that the generator's steel cabinet would act as a reasonably good Faraday Cage. As my generator is less than three feet from my heated garage, which already had ethernet running to it, it wasn't difficult to do. If you want to use Wi-Fi, and your network router is close enough, the enclosure I used is weather-proof, and could be mounted on the outside of the generator. You'd just have the run the power/serial cable into the generator. 

BTW, when I drilled through my **new** generator's cabinet to install my monitor, I found that the cabinet is aluminum. Unfortunately the spot I picked also has a steel inner liner. The two layers combined are easily 1/8". It was a bugger to find a grommet that would work.

karamanos1, 2023-09-10:
Definitely a learning experience/curve for me with this stuff. You wouldn't
be willing to build one for me. I'd be willing. LOL I live in Maryland.

lwbeam, 2023-09-10:
Sorry, no.

karamanos1, 2023-09-10:
Thanks for considering. Lol I already ordered the parts. Wish me luck. I may send you a few messages if I get stuck. Thanks again.