jgyates/genmon #829: genmon test:  No module named 'cryptography'

dmperl, 2022-12-12:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

genmon successful startup

### Actual Behavior

```
pi@genmon:~/genmon $ sudo python genmon.py

This program requires the modules located in the genmonlib directory in the github repository.
Please see the project documentation at https://github.com/jgyates/genmon.

Error: No module named 'cryptography'
pi@genmon:~/genmon $
```

### Steps to Reproduce (including precondition)

Follow manual install instructions in section 3.4
https://github.com/jgyates/genmon/wiki/3.4--Manually-installing-genmon-software-(not-using-the-setup-script)

Note this step failed:   sudo apt-get install python-serial
Replaced with (success): sudo pip3 install pyserial

Also tried: bash genmonmaint.sh -i  (section 3.3)
errors:  can't find rust compiler requires 1.48.0.   Failed to build cryptography.  Reverted to manual install to help with debug.

Have successfully completed the serial test:
```
pi@genmon:~/genmon $ cd Oth*
pi@genmon:~/genmon/OtherApps $ python3 serialtest.py

Note: Genmon must NOT be running for this test to work properly. If Genmon is running this test will not function properly,

Loopback testing for serial port /dev/serial0...

Using python serial library V3.5
Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful
pi@genmon:~/genmon/OtherApps $
```

Flask 2.2.2 (haven't been able to test webserver yet)

Also when using script setup https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software
It is also recommended to update your system software first with the following commands:

```
 sudo apt-get update
 sudo apt-get distro-upgrade  --->  should be  --->  sudo apt-get dist-upgrade
```

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: Generac Evolution  Firmware V1.21, Hardware V1.06
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.14

jgyates, 2022-12-12:
If this command failed:

   bash ./genmonmaint.sh -i 

then your errors are very much expected.

What operating system are you running? Is it the 32 or 64 bit version of the Raspberry Pi OS?

dmperl, 2022-12-12:
Raspberry Pi 2B,  
Raspberry Pi OS Lite (32 bit) Bullseye 5.15.61-v7+

Thanks!

jgyates, 2022-12-12:
I updated the wiki to fix the type for "dist-upgrade" typo. I will update the manual install or remove that page.

the manual instructions are out of date. If you want to manually install then I would reference the script genmonmain.sh and the function installgenmon(), however typically issues of the install script failing is due to mirror failures, which I can't really control. One of the issues with pip is that once a mirror is selected, if there is a failure with that mirror then there is no fallback to try a different mirror.

Try reinstalling with the genmonmaint.sh script and post your errors if you have any and we can go from there.

dmperl, 2022-12-12:
Knowing that I have a good serial connection, I'll start fresh with a new image tomorrow and forge ahead with: genmonmaint.sh -i

thanks!

jgyates, 2022-12-12:
OK. I am going to close this issue but feel free to post related questions to this thread.

closed 2022-12-12 by jgyates

jgyates, 2022-12-13:
Try running this:

    pip3 install --upgrade pip

Then run the install again.

jgyates, 2022-12-13:
try this:

    sudo pip3 install --upgrade pip

jgyates, 2022-12-13:
also, according to your log it looks like everything installed OK except these two:

     sudo pip3 install pyopenssl
     sudo pip3 install fluids

after you run "sudo pip3 install --upgrade pip" try installing the two packages above

dmperl, 2022-12-13:
deleted pip3 from /home/pi/,local/bin  then re-did as:
> sudo pip3 install --upgrade pip

[image]

success for this step.

Next:
> sudo pip3 install pyopenssl

[image]

Complete console log attached: 
[20221213 sudo pip3 install pyopenssl COMLETE console log.pdf](https://github.com/jgyates/genmon/files/10219333/20221213.sudo.pip3.install.pyopenssl.COMLETE.console.log.pdf)

This is "it", but what next?  ;-)   Thanks!

jgyates, 2022-12-13:
couple of other things to try (install cargo first, then install cryptography):

    sudo apt-get -yqq install build-essential libssl-dev libffi-dev python3-dev cargo
    sudo pip3 install cryptography
    sudo pip3 install pyopenssl

dmperl, 2022-12-13:
Cryptography install (success with warning):
[image]

pyopenssl fail, see attached:
[20221213-2 sudo pip3 install pyopenssl COMPLETE console log.pdf](https://github.com/jgyates/genmon/files/10220031/20221213-2.sudo.pip3.install.pyopenssl.COMPLETE.console.log.pdf)

I'll circle back and try the build-essential recommendation.

jgyates, 2022-12-13:
The warning is expected when you run "sudo"

jgyates, 2022-12-13:
let me know how the build essentials works. I think that is the one to go with.

also, your last log, it looks like it is pulling cryptography V38 from your cache instead of using the installed version V37. If the build essentials does not work then you can remove the V38 from the cache with this command:
  

     sudo pip3 cache remove cryptography

dmperl, 2022-12-13:
Cryptography re-install (success with expected warning warning):
[image]

pyopenssl  SUCCESS!
[image]

fluids SUCCESS!
[image]

back later to test genmon

dmperl, 2022-12-14:
All good.   Next, configure for Home Assistant.  THANKS!!!

If you are making any changes to include:
> sudo apt-get -yqq install build-essential libssl-dev libffi-dev python3-dev cargo

I'd be happy to test a fresh install  (on another SD card  ;-)

jgyates, 2022-12-14:
That is great. The updated genmonmaint.sh is already checked into the repository. Looks like the fix was to just add cargo. The script was already doing the rest. Thanks for all your testing and let me know if you have any other issues.

dmperl, 2022-12-17:
I did a fresh install (specifying ssh public-key authentication only)  on a different microSD card.
The System Update portion of the install ran without error. ssh log attached:
[putty 3.3 Setup genmon software System Update.log](https://github.com/jgyates/genmon/files/10252528/putty.3.3.Setup.genmon.software.System.Update.log)

The Genmon Install threw out a few errors, but in the end it all worked. ssh log attached (search for error)
[putty 3.3 Setup genmon software Genmon Install.log](https://github.com/jgyates/genmon/files/10252531/putty.3.3.Setup.genmon.software.Genmon.Install.log)

I restored the config files from my original installation (4 days ago) using PSCP, which leads me to my first "enhancement" suggestion,   On the **About** page, there is a **Backup** button to download backup files.  A **Restore** button would be nice to have.  ;-)

THANKS!!!!!!!!!!!!!!!!!!!!!

jgyates, 2022-12-18:
I will add that to the feature list

jarocht, 2023-06-02:
Having similar issues. pyopenssl is erroring on the wrong version of rust (1.48.0 and expecting 1.56.0).
Have tried the:
sudo apt-get -yqq install build-essential libssl-dev libffi-dev python3-dev cargo

but still getting the same error when installing cryptography or openssl.

I'm running this on rasbian lite, on a pi zero w.

jarocht, 2023-06-02:
Spent some more time playing. This seems to be a version dependency issue.
pyopenssl versions now require Crypto v41+, and crypto v41+ is ignoring the 'don't build rust flag' and trying to build with an outdated version of rust.

I manually installed crypto v37.0.4 and pyopenssl v22.1.0

I attempted to update cargo/rust but was unsuccessful and started hitting other issues.

restarting the genmon service I'm now getting an illegal instruction error.... off to research that!

lwbeam, 2023-08-20:
Just want to confirm that this is an on-going issue. I've recent had to replace my generator and am now in the process of transferring my Genmon setup to the new machine. As part of this process I decided to do a clean install of Genmon on a new SD card using the latest version of Raspberry Pi OS (32-bit Bullseye Lite). I spent 6 hours trying to get the latest versions of rust, pyopenssl and cryptography installed, so that the Genmon install would complete successfully. Eventually I gave up and went to bed. The next day, without seeing this post, I decided to try installing Genmon using the 64-bit version of the Raspberry Pi OS. It installed immediately without issue. I get that this isn't an issue with the Genmon installation routine per say. Just want to add a datum point.

jgyates, 2026-04-05:
@dmperl, earlier you asked for a restore option in the UI. This has been added with 2.0