jgyates/genmon #830: Update failure V1.18.15 to V1.18.16

DJT1423, 2022-12-17:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Attempting update from V1.18.15 to V1.18.16.  Update fails.

### Actual Behavior

Update downloads and seems to process, GenMon restarts and I log back in, but it continues to show V1.18.15 and version V1.18.16 available. Pi has been updated and rebooted. Have repeated the update process 4 times.

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

Pi 3B+; Generac 11KW natural gas;

- Generator Model: Generac 11KW natural gas
- Genmon Logs: Submit via the About page
- Genmon Version: V1.18.15
[genmon_logs.tar.gz](https://github.com/jgyates/genmon/files/10252093/genmon_logs.tar.gz)

skipfire, 2022-12-17:
Have you edited any files in the genmon folder?  If you log into the Pi, `cd genmon` and then run `git status` it will show you if there are modified files and if there's a conflict that could block getting the current code.

DJT1423, 2022-12-17:
> Have you edited any files in the genmon folder? If you log into the Pi, `cd genmon` and then run `git status` it will show you if there are modified files and if there's a conflict that could block getting the current code.

No, I haven't modified anything, not knowingly. Attached is an image of the results of 'git status'.
[image]

jgyates, 2022-12-17:
git status shows that you are up to date.

restart genmon, refresh your browsers and send a screen shot of the version information on the about page.

Also, from the genmon folder you can type this command:

     cat ./genmonlib/program_defaults.py 

This file will contain the version of genmon.

You can also show what is happening when genmon restarts by typing this command:

     ./startgenmon.sh restart

I also noticed that you are running buster (Debian 10) and not bullseye (Debian 11). It is possible that your apt repository conf files need to be updated. Type this command:

     sudo apt-get update

and post the results.

If this works without errors, then update your system software

jgyates, 2022-12-17:
also, to manually update the software, from the genmon folder type this:

     ./genmonmaint.sh -u

Then type 'y' when prompted.

DJT1423, 2022-12-17:
Success!. Thank you. Not sure what did it, but I'm happy. Screen shots attached of each step. Thanks, David
[image]
[image]
[image]
[image]
[image]

jgyates, 2022-12-17:
This is just guess as to why this occurred: About 3 days ago I added a line to genloader.py that would automatically update the apt sources file:

     sudo apt-get -yqq --allow-releaseinfo-change update

I added "----allow-releaseinfo-change"

I noticed that some people who had submitted logs had part of their updates broken because the Raspberry pi foundation updated the older butster repositories to "oldstable" instead of "stable". This change requires an update to the file "/etc/apt/sources.list". The new "--allow-releaseinfo-change" does that.

I am guessing that you had this older repository listed as "stable" or "testing" in your sources file.  Updates will not work until the /etc/apt/sources.list file is updated.

closed 2022-12-17 by jgyates