jgyates/genmon #831: json code

andrewvanlagen, 2022-12-20:
hi 
Where in the code is the gui_status_json and the status_json configured 
thanks

jgyates, 2022-12-20:
I will answer this question assuming an Evolution/Nexus controller.

status_json is in the function DisplayStatus in /genmon/genmonlib/generac_evolution.py

https://github.com/jgyates/genmon/blob/64eeddafbad42b7bd2461f323303dbd85c12bf57/genmonlib/generac_evolution.py#L4684

the gui_status_json is in the function GetStatusForGUI in the same file:

https://github.com/jgyates/genmon/blob/64eeddafbad42b7bd2461f323303dbd85c12bf57/genmonlib/generac_evolution.py#L4841

If you want another controller then look for the same function names in generac_HPanel.py, generac_powerzone.py or custom_controller.py

both of these functions return a python dictionary which is converted to JSON here:

https://github.com/jgyates/genmon/blob/64eeddafbad42b7bd2461f323303dbd85c12bf57/genmon.py#L919

Let me know if you have any other questions.

jgyates, 2022-12-20:
I am going to close this issue but feel free to post to this thread with any follow up.

closed 2022-12-20 by jgyates