jgyates/genmon #832: Voip.Ms

andrewvanlagen, 2022-12-22:
i am getting error in my logs

2022-12-21 21:38:04,970 : Error in MyVoipMs:SendSMS: This IP is not enabled for API use : myvoipms.py:51
2022-12-21 21:38:04,971 : Error sending message in QueueWorker, callback failed, retrying

what ip has to be set in voip.ms

jgyates, 2022-12-22:
In your voip.ms account both your main accound and and sub account must not have IP Address restrictions enabled. Alternatively you could add your WAN IP address that your Raspberry Pi is located.

For the main account look under Main Menu -> Account Settings ->  Security -> IP Restrictions 

This is off by default.

For and sub accounts go to Sub Accounts -> Manage Sub Accounts, the select the sub account and go to Security->IP restrictions.

These are also off by default.

andrewvanlagen, 2022-12-22:
I enter my wan Ip in ip restrictions still doesn't work

Do I have to enter a ip in api configuration setting

jgyates, 2022-12-22:
actually go to Main Menu - > SOAP and REST/JSON API 

This is the IP address for the API, not VoIP calling. Here you can enter you IP address along with your API password.

You can get the WAN IP address for the IP by typing this at the console:

    curl http://ipinfo.io/json

jgyates, 2022-12-22:
I will add this info to the wiki

andrewvanlagen, 2022-12-23:
thanks that worked