jgyates/genmon #833: Feature request: clear maintenance alarms

aa777888-2, 2022-12-26:
Controller: Nexux, Air Cooled.

Feature request: is there anything in the Nexus interface that will/would allow genmon to reset maintenance alarms (e.g. check battery, change oil, etc.)?

jgyates, 2022-12-26:
@aa777888-2 

That is a great suggestion, however I don't know the modbus command to archive that functionality.  It is possibly that the Nexus firmware does not support this functionality. I have tried issuing the same commands to do this for Evolution on the Nexus, but it does not work when I did, however you are welcome to try it again as this may be dependent on firmware versions. You can follow the description in this thread to test it:

https://github.com/jgyates/genmon/issues/827

if you do decide to test this and it works let me know and I can add the buttons to the web interface.

I am going to close this thread however please post to this thread for any follow up status or additional questions on this topic.

closed 2022-12-26 by jgyates

aa777888-2, 2022-12-26:
Thanks for the quick reply! I was motivated to ask because we just had some extended utility outages here, 46 hours over the last three weeks, and during that time the LCD display failed on the controller and then the generator threw a maintenance alarm, the alarm being nicely visible thanks to genmon :-) but with no easy way to clear it.

Right before I went out to attempt to repair the LCD display I posted the request. I was victorious in my repair activities. Typical Generac electronics failure: build-up of corrosion leading to shorted pins on the display. Corrosion removed, functionality repaired, alarms cleared. However, the referenced issue #827 is quite interesting and I will be sure try that method the next time I have an alarm to clear. Thanks again!

grsthegreat, 2022-12-28:
im confused. is there a way to remotely clear alerts and alarms on the evolution generators?

jgyates, 2022-12-28:
Yes and No.

For clearing most alarms remotely:
Evolution Liquid Cooled Tested and validated. Functionality in the Web UI for this model.
Evolution 2.0 air cooled - tested and validated with firmware V1.15. This functionally in the Web UI for this model
Evolution 1.0 air cooled - When this test was reported to me it did not appear to work. This functionality is NOT in the Web UI 
Nexus Air Cooled- When this test was reported to me it did not appear to work. This functionality is NOT in the Web UI 
Nexus Liquid Cooled - When this test was reported to me it did not appear to work. This functionality is NOT in the Web UI 

If anyone wants to retest Evo 1.0 or Nexus Air or Liquid cooled the method to to this and the remote on/off/auto commands are in this thread: https://github.com/jgyates/genmon/issues/827

Once validated I can enable the button in the UI to do this.

One thing to keep in mind is that some alarm conditions are not remotely clearable. Things like maintenance alarms and other non-critical alarms are so it is possible that previous tests were not testing a valid situation for clearning the alarm.

ripvega, 2022-12-28:
> Once validated I can enable the button in the UI to do this

Will you be able to add MQTT support for this button too, please?

jgyates, 2022-12-28:
The buttons in the UI are not represented via MQTT, however all commands that you can issue with ClientInterface.py can be issued via MQTT with the /generator/command topic. for example:

publish:
        topic: "generator/command"
        payload : "setremote=resetalarm"

more info is here:

https://github.com/jgyates/genmon/wiki/1----Software-Overview#genmqttpy-optional

all base commands are defined here:

https://github.com/jgyates/genmon/blob/9e5b2ed58673b191d45874a892218ee2f40f4ffc/genmon.py#L781