jgyates/genmon #834: Plugging genmon into a live Nexus Air Cooled controller causes Nexus controller LCD display to go blank

aa777888-2, 2022-12-28:
### Expected Behavior

Plugging genmon into a powered-up Nexus Air Cooled controller does not disturb operation of the controller.

### Actual Behavior

Plugging genmon into a Nexus Air Cooled controller that is powered up will occasionally cause the LCD display on the controller to go blank. It will remain that way until power is cycled on the controller by disconnecting/reconnecting the two wire 120VAC power connector. Other than a blank LCD screen the controller appears to continue to operate normally.

### Steps to Reproduce (including precondition)

Plug a genmon unit into the controller accessory port. Repeat as necessary to cause the problem. This need not be done/probably should not be done rapidly.

It should be noted that my genmon is a Raspberry Pi Zero 2 W and it is powered directly from the Nexus Air Cooled controller accessory port 5VDC pin.

### Screenshot or Pictures relating to the problem (if possible)

None applicable.

### Your Environment

- Generator Model: 0060531 (as reported by genmon), 17KW air cooled.
- Controller Model: Nexus Air Cooled Assembly 0H6680D REV J C1214 Software 0J3401 REV C.
- Genmon Logs: not applicable.
- Genmon Version: 1.18.16.

### Detailed Discussion

I don't think this is a genmon software problem per se, but since this is the repository of all genmon knowledge and experience it seemed that my experiences and observations should be posted here.

Recently I had the Pi crash on me. I did not perform any due diligence on this issue. It's probably a Raspbian problem of some sort. It's not the first time. I see about one crash every 6 months or so. The only way to recover from the crash is to power cycle the Pi. Since my genmon obtains its power from the Nexus Air Cooled accessory port 5V pin, the only way to do this is to unplug it, wait a few seconds, then plug it back in again. This has never been a problem in the past. However, most recently, this caused the LCD on the Nexus controller to go blank.

Initially I thought this was due to some internal corrosion on the LCD connections inside the controller. However, today, in making another repair to my generator, I found that the problem is actually associated with plugging the genmon into a live controller. This is repeatable although it sometimes takes several attempts. It can be overcome by power cycling the controller by means of disconnecting/reconnecting the 120VAC two wire power connector.

### I am wondering: has anyone else experienced this problem?

jgyates, 2022-12-28:
I will let others with a Nexus chime in on this, however you may want to consider powering the pi off the battery instead of the accessory port. The startup current of the pi may be reaching the limit of the accessory port 5V supply. While your pi is running you may want to check your pi's power stats on the Monitor page (Freq throttling, Freq Cap and Undervoltage). These can be an indication that the pi is not getting a reliable source of power.

aa777888-2, 2022-12-28:
On the monitor page CPU Frequency Throttling, ARM Frequency Cap and Undervoltage all report "OK".

The Zero 2 W is a pretty small load. The largest numbers I've seen quoted with a heavily active CPU and Wi-Fi are around 600mA at 5V. It seems unlikely it's working that hard to run genmon. But perhaps it's riding the ragged edge, and perhaps that's why it is crashing every few months.

In the future I'll be restarting by cycling Nexus power. If it becomes an issue then maybe I'll power it differently. I had originally planned to use PoE but the idiots who installed the generator stuffed the wire conduit out to the generator too tightly to get another wire through it.

jgyates, 2022-12-29:
I searched the issues and found these:

https://github.com/jgyates/genmon/issues/117

In this thread https://github.com/jgyates/genmon/issues/31

@danorum stated:

"I wanted to follow up. I was powering my Pi Zero (with USB ethernet dongle) from the molex connector. When I plugged in the monitor to the generator, the LCD display went blank. I had to reset the controller to get the display back. This reproduced 100% for me. I pulled the red pin from the molex connector and used an external power supply. (I kept the black wire connected to have a common ground.) In this configuration, genmon works AND the LCD stays on. It looks like I need to use the battery power supply instead of the controller molex to supply power to the Pi."

jgyates, 2022-12-29:
I wonder if this is specific to the pi zero and Nexus interaction or does this occur with any pi variant? I know lots of folks are using Nexus with other pi's and this has not been reported with anything but the pi zero.

aa777888-2, 2022-12-29:
First off, that is above and beyond, jgy, doing the search I ought to have done. Thank you!

As you write, It is interesting that the only two reports involve a Zero and a Zero 2 W. Both are lower power variants so one would think those are going to be a better choice in terms of power consumption. In one case you have the added burden of the USB NIC, in the other on-board Wi-Fi, but in both instances it should still be a much more lenient case power-wise than attaching a Pi 3 or 4 variant.

It could simply be a transient condition issue. In my admittedly limited testing (no point in stressing the systems and causing worse problems), power cycling the Nexus works 100% whereas re-plugging the genmon causes problems maybe every fourth cycle. If that's the case then I can live with the issue because it's a rare occasion that the genmon needs a power cycle and cycling the Nexus+genmon is no more difficult than cycling the genmon by itself.

However, there still is the bigger question of why I'm seeing a Raspbian crash of some sort every few months. It could be a power issue, or it could be something else. I've been kind of blowing it off as annoying but not critical. The last time it happened a casual look at the logs did not immediately show me anything about what failed. However it did show that the Pi did keep running after whatever happened happened. Things definitely do go weird in the OS, though, and all Wi-Fi comm's go dead, so it's very difficult to debug.

jgyates, 2023-01-03:
I am not sure about the life or age of your SD card. It could be going bad. Make sure it is rated for a high number or writes (class 10 SD card is recommended). I also use a ram disk for the logs to extend the life of the SD card:

https://github.com/jgyates/genmon/wiki/Appendix-M-Tips-and-Tricks-useful-for-Genmon#extending-the-life-of-your-sd-card

I am closing this issue but feel free to post here if you have any other data or questions on this topic.

closed 2023-01-03 by jgyates

aa777888-2, 2023-02-05:
**Bottom line up front:**

Following up on the "Pi crashing occasionally problem".

It's not related to sourcing power from the Nexus but apparently due to a bug in the onboard WiFi firmware.

**Detailed discussion:**

I totally rebuilt the system onto a new SD card, this time starting with the latest Raspian 64 bit for the Zero 2 W.

The Pi remains powered by the Nexus controller.

Occasional crashing continued through the month of January. The crashes will take the device off WiFi for a while, then cause a spontaneous reboot, or require an outright power cycle to reboot.

I have finally tracked down the root cause to this known issue with Zero 2 W onboard WiFi stability:

[https://github.com/raspberrypi/firmware/issues/1768](url)

TLDR: current thinking is that this bug _might_ have something to do with roaming across multiple WiFi access points.

My WiFi environment consists of 4 Unifi access points all carrying the same three SSIDs. One SSID is dedicated to IoT devices and the genmon is on the IoT SSID.

While waiting for a real fix, I've used the Unifi system to lock the genmon Zero 2 W to the nearest, single access point. It'll be at least a couple of weeks before it can be determined if this had a positive effect on the problem or not.

jgyates, 2023-02-05:
Great info. Thanks for the detailed follow up.

aa777888-2, 2023-02-15:
Locking the Z2W to a single AP has reduced spontaneous reboots from approx. once per day to about twice per week. Still not fixed, though.