jgyates/genmon #835: Question about Genmon and remotely controlling the Generator via contact closure (relay)

alexsgocart, 2023-01-02:
Hello,

I recently got a Generac Air Cooled 22kw Generator and got Genmon all set up. One of the things I saw in the documentation was a way to sent GPIO inputs to the Pi to command the generator for start/start+transfer/stop. 

As the system is hooked up, I have four SMA Sunny Islands with batteries that power my home during an outage. The generator is there to charge the batteries when solar is down (rain storm/night time). The nice part about the Sunny Islands is it uses solar power to charge the batteries when there is sun (and isolates itself from the grid automatically).

How I currently have it set up is the Sunny Islands will control the power in the home. The grid power comes in and goes to the Generac transfer switch. That is where the generator is also hooked up. That goes to the Sunny Island and there is a relay that tells the Sunny Island is the power grid or generator.

The Sunny Island has the ability to command the generator with an internal contact closure. If it requests the generator, it will close one of its relays to call for the generator. When the generator is no longer needed, it will safely disconnect inside the Sunny Island, wait a few seconds, then turn the internal contact off (open).

Now what my question is with Genmon:

Is it possible to have Genmon be commanded by a single GPIO pin? I am newer to Raspberry Pi's but if I am thinking this correctly, it would go: Power/Ground from the Pi > to one side of the Sunny Island relay > out the other side of the Sunny Island relay to a resistor (from what I can find, 10k ohm resistor should be inline with this?) > to the GPIO pin on the Pi?

With this intended setup, when the Sunny Island requests the generator, it will close the contact closure which then would command the Pi to start the generator (yes it would go through it's proper start sequence like warm up time and stabilize voltage/rpm), which the generator transfers the transfer switch, which sends the electricity to the Sunny Island. The Sunny Island waits to make sure the power is stable and switches over.

When the Sunny Island is finished with the generator, it will disconnect its internal transfer switch to stop accepting the generator power, wait for a short period of time, then open the contact closure, which then the Pi will see that GPIO pin circuit is open and initiate the stop command.

Right now as a test, I tried the 'Genmon GPIO Inputs' found in the Web UI and followed the gengpioin.py wiki page.

As an experiment, I tried Pi 3.3v power to the contact closure in the Sunny Island > out the Sunny Island to a 10k resistor > then to GPIO 27 that I assumed would trigger the generator to start it. The problem I ran into is, when I set this up, the generator endlessly short cycles (with the contact closure open and closed on the Sunny Island), which I stopped experimenting with and is why I am here because I don't want to ruin the new generator. I tried falling/rising/both/up/down/off and no success and just starts the generator, runs for 5-10 seconds, goes into cooldown mode for about 30 seconds, shuts down, repeats.

Any help or suggestions is highly appreciated. 

If a picture of how everything is wired currently is helpful, I can draw one out to explain how everything is wired.

If I'm way over my expectations of a small cool project, I will just go with the 2-wire method for the generator. I just wanted to learn something new and experiment with this to add it to my home automation as I am learning ESPs and Raspberry Pi's.

Thanks.

- Generator Model: Air Cooled 22kw
- Genmon Logs: I can submit the logs if requested (I don't think they are needed?)
- Genmon Version: V1.18.16

ripvega, 2023-01-02:
If you look through the generator documentation/wiring diagram (assuming
you have an Evolution controller), there is an option called, from memory,
2-wire start, which may do what you want.

skipfire, 2023-01-02:
@alexsgocart have you tried without the resistor?  If you connect 3v3 and GPIO to the relay you will want rising edge and pull-down, I'm not sure about the callbacks option, that is also specified twice in the docs of the config file and I have no clue if that would cause misbehavior if it is also defined twice in the real config.  Do you have anything connected to GPIO17?

The next problem is that a single relay may not do everything you need, these are edge triggers so a single pulse is all that is needed to initiate the command, but you would need an additional pulse somewhere to initiate the stop command.  You will want a transfer start once you get everything worked out (GPIO22).

jgyates, 2023-01-03:
@alexsgocart 

I agree with @ripvega. Two wire mode was made for this type of thing and it bypasses the pi so it removes it as a point of failure for emergency operation. Currently a pi running genmon is simply monitoring the generator and relaying any problems so a service outage can be avoided. Both the pi and genmon are not designed to act as the center point of an outage response. Also note that when genmon issues a start, stop or start/transfer command, these are just commands that are sent to the controller firmware and do not take priority over other controller operations. These commands exist to test these features of the firmware and the controller does not treat these commands with priority. For example if you are running in two wire mode, I believe these commands will not work. Also if you are currently running an exercise cycle they may not work. I have not validated myself, however these are commands that I have found, but to my knowledge Generac does not advertise these features or ship products that use them.

In short, can you you do this? Probably, yes. But **should** you design your system this way? My opinion would be no, use two wire mode that is built in the controller.

I am going to close this thread but feel free to post follow up questions or status updates. Let us know what you decide and how it works out. It is always good to hear folks using genmon and generators in non typical applications.

closed 2023-01-03 by jgyates

alexsgocart, 2023-01-03:
Thank you for all your suggestions and help.

I will continue experimenting and see what I can come up with. I totally understand the risks that come along with relaying on the potential issues of the Pi/Genmon coming into play under emergency situations and have taken that into account. I will take all the suggestions here and try them, but most likely will just end up using the 2-wire way.

If I figure out a way that works, I will post an update here on how I got it all working together as the Sunny Islands are older and the Sunny Webbox is very hard to come by and is ancient in today's standards of home automation.