jgyates/genmon #836: Generator Command Response email looping

T-C123, 2023-01-02:
I’ll start by saying that this is my first experience with Raspberry Pi, Linux, and Python. My programming days were many years ago and, although I can follow a recipe, I am not a chef. I came across some problems setting up the Genmon email command system and thought my experience and solution might help others. 

**Expected Behavior**

Genmon monitors the “Generator” folder and responds to emails with “Generator:” in the subject line.
Genmon then processes the command and moves the message to the processed folder.

**Actual Behavior**

Message is processed.
Reply message is sent with the subject “Generator Command Response…. “

Genmon seems to read “Generator Command Response….“ as another request and sends an error message with the subject “Error in Generator Command (command prefix)”. If I’m not careful, this will rapidly continue to loop until a limit of 500 messages are sent and Gmail locks me out for a day. 

**Steps to Reproduce** 

This happens when I filter only on the subject containing “Generator: “

**My solution**

I changed the filter to include both my recipient email address AND subject includes “Generator: “

**Suggestion**

I believe the problem is that Gmail saw the outgoing Genmon response contained the word “Generator” , ignored the need for a colon, and processed it adding the label “Generator”. (Gmail uses Labels instead of different Folders.) I think Genmon then sees its own response with the Generator Label and the subject “Generator Command Response…. “ and processes it as a new (but incorrectly formatted) request sending an error message. Gmail then sees that this error message contains the word “Generator” in the subject, labels it “Generator” and Genmon again reevaluates, sending endless error responses. 

Creating a very specific mail filter helps prevent this. 

Changing Gammon’s response to NOT include the word “Generator” in the subject line (maybe something like “Genmon Command Response”) or allowing the user to create a very unique Subject for Genmon to look for and act on (like MyGen456) instead of “Generator:” would be a big help.

**Screenshot or Pictures relating to the problem** (if possible)

Notification email inbox:

[image]

Genmon monitored mailbox:

[image]

**Your Environment**

Generator Model: Generac 18kw Evo2
Raspberry Pi 3 B+
PintSize.Me HAT w/fan
Home brew fan control and oil temperature monitoring
Genmon & Raspberry updated to latest versions
Dedicated Gmail account for Genmon with two factor authentication and an app specific password

jgyates, 2023-01-02:
@T-C123 

Yes, this would be the correct approach.

this page:

https://github.com/jgyates/genmon/wiki/1----Software-Overview

has this info:

A note about email and security: The genmon.py app monitors folder on an IMAP email account for emails with a subject containing "generator:". If this is found, the remaining characters of the subject filed are parsed for commands for the monitor program. Since commands are sent to genmon.py via the subject line of an email, is is suggested that some level of email filtering occur on the receiving email account. For example, gmail supports email filters that if emails arrive from specific people with specific words in the subject, move them to specific folders. You could then have genmon.py monitor the specific folder used in your filter to only allow specific emails to send commands to genmon.py. If this approach is used then genmon.py can be configured (via mymail.com) to monitor a specific IMAP folder."

Genmon will read **any** email in the specified incoming folder. That is by design. Commands are filtered based on key word so the behavior you are seeing is expected. You just needed to update your filter in gmail. To make this a little more forgiving I will change the response from "Generator Command Response" to "Genmon Command Response". 

thanks for the feedback. Let me know if you have any other issues or questions.

closed 2023-01-02 by jgyates

T-C123, 2023-01-02:
Thank you for the fast response!

jgyates, 2023-01-03:
I checked in an update that changes the response to "Genmon Command Response".

T-C123, 2023-01-03:
Great! Thank you very much!
Tom