jgyates/genmon #837: Minor issue - Wifi signal strength appears inconsistent between generator status page and monitor page

DC072012, 2023-01-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

WIFI Signal strength on status page and the monitor page should be similar if not equal

### Actual Behavior

Signal strength on status page and monitor page are not similar.  Coincidentally they appear to add up to roughly 100.

I tried this on both Safari and Chrome so it does not appear to be a browser issue.

Maybe this in unavoidable for some reason - I could not find a similar issue when searching the wiki.

### Steps to Reproduce (including precondition)

Click on status screen, then the monitor screen

### Screenshot or Pictures relating to the problem (if possible)

3 screenshots:
[image]
[image]
[image]

### Your Environment

- Generator Model: [{Please write here}](http://192.168.57.170:7500)
- Genmon Logs: Sent
- Genmon Version: V1.18.16

jgyates, 2023-01-02:
This is an anomaly / bug. I will try to correct this behavior soon. The software is pulling the info form 2 different places but they are related as you noted.

jgyates, 2023-01-03:
I have updated the code that **should** correct this. There are still some rare conditions that this can occur, but that is dependent on the underlying wifi network adapter device driver. Not all drivers support the same format in /proc/net/wireless and if the program "iw" is not present then the software falls back to using /proc/net/wireless to get the signal strength. I am going to close this issue but let me know if you have any issues in this thread.

closed 2023-01-03 by jgyates

DC072012, 2023-01-03:
Looks like the update is working!  Thanks for the quick update.