jgyates/genmon #838: Login Configuration

DJT1423, 2023-01-02:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

I have messed up my login configuration and can no longer access the web portal for my generator. I was using no SSL and logging in locally via IPAddress:8000. I recently added a Let's Encrypt certificate to the device I was using (Raspberry Pi 3B+), and thought perhaps I could use that on GenMon. I ticked to use HTTPS left self-signed unticked, and added the file paths for the key and certificate file and saved the changed. I could not then access Genmon.

I have gone in the back door to edit the genmon/conf/genmon.conf file - I changed "usehttps" to False and rstopped/restarted the monitor, but still am unable to access GenMon.

Can you provide some guidance please?  I have attached a copy of the config file (with a .txt suffix replacing .conf).
Thanks

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}
[genmon.txt](https://github.com/jgyates/genmon/files/10333719/genmon.txt)

skipfire, 2023-01-02:
@DJT1423 you said you restarted the monitor, was that with startgenmon restart or some other command?  Have you tried just rebooting to make sure nothing is being held in memory for the old configuration?

DJT1423, 2023-01-02:
I stopped and re-started Genmon with these commands in terminal:
cd /home/pi/genmon
./startgenmon.sh stop
./startgenmon.sh start

I have now re-booted the machine but am still unable to access the GenMon webpage

David

skipfire, 2023-01-02:
Have you checked the startup logs?  `cat /var/log/genserv.log` should show how the website starts up.  The entry for mine is 
`Starting /home/piadmin/genmon/genserv.py, Port:8000, Secure HTTP: True, SelfSignedCert: True`

DJT1423, 2023-01-03:
I checked that log file - this looks interesting, but unfortunately does not mean anything to me:

2023-01-02 19:08:15,622 : Starting /home/pi/genmon/genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True
2023-01-02 19:08:15,903 : Error in app.run: [Errno 98] Address already in use : genserv.py:4273

David

David

***@***.***

skipfire, 2023-01-03:
That says it is still trying to start with HTTPS on port 443, you might try accessing `https://{ipaddress}/`

DJT1423, 2023-01-03:
I need to use a different port than 443, another function is accessed there.
I have stopped the service, set the genmon.conf to usehttps = False; use sefsignedcert = False; the https port to 8443 and the http port to 8000, and then rebooted the Pi.
The startup log however seems to say my changes (repeated multiple times) are being ignored:
2023-01-02 19:31:57,312 : Starting /home/pi/genmon/genserv.py, Port:443, Secure HTTP: True, SelfSignedCert: True

However, when I review the genmon.conf file, the changes are reflected there. It seems they are being ignored?

David

***@***.***

jgyates, 2023-01-03:
I would recommend starting with a fresh copy of genmon.conf. You can do this from the command line with this command:

     cd genmon
     sudo cp ./conf/genmon.conf /etc/genmon

Then restart genmon:

    ./startgenmon.sh restart

Since you got this error in your logs:

    Address already in use

you may consider rebooting. This error is caused by multiple copies of genmon running at the same time. If you continue to see this after a reboot then I would check your cron entries that load genmon on boot.

Let me know if that does not solve your issue. I am going to close this issue but feel free to post to this thread with any follow up questions or status.

closed 2023-01-03 by jgyates