jgyates/genmon #839: Auto reset of remaining fuel

ripvega, 2023-01-02:
I use 2 x 47kg LPG bottles, which equates to 192 litres.

When the automatic changeover valve swaps from an empty service pair of bottles to the full reserve pair of bottles, I'd like the estimated remaining fuel to reset to 192 litres. I have a sensor to detect the change and this already shows up in the Monitor as an external data input.

What would be the best way to do this?

ripvega, 2023-01-03:
Great, that works. Thank you.

For testing purposes, without changing bottles and running the generator, is there a way to set the estimated remaining fuel manually so that I can then test the reset?

ripvega, 2023-01-03:
I've been having more of a dig with this and I think that

http://10.0.1.233:8000/cmd/fuel_log_clear

should probably do what I want, but it returns "Power Log is empty".

Is that correct?

jgyates, 2023-01-03:
No, set_tank_data tells genmon the percentage of fuel in the tank for the UI gauge and estimating the time left with the fuel i the tank. It is not related to the power log or fuel log.

Genmon will read the power in watts or amps when the generator is running. It will keep a log of this data. This is the power log. 

Fuel log is an optional log that tracks the fuel level over a period of time. fuel_log_clear will clear this, if the option is enabled. Fuel log is off by default and only enabled if you set "Log Fuel Level to File" on the advanced page.  This has nothing to do with reseting the amount of remaining fuel.

power_log_clear will clear the power log. The power log is used to calculate the remaining fuel in the tank. The power log tracks how long the generator runs and how much power (in watts) it uses. This data, along with the fuel consumption curves for your specific generator calculate how much fuel is used over time. The idea is that the tank is assumed to be full and the power log lets you calculate how much fuel has been used.

The advanced page has a value called "Subtract Fuel" that allows you to subtract an amount of fule from the tank. For example if you have a 100 gallon tank and your power log is empty then genmon assumes you have 100 gallons of fuel. If "Subtract Fuel" is set to 10, then you would only have 90 gallons with the power log empty.

If I understand what you are wanting, you only need clear_power_log and "Subtract Fuel".

Let me know if you have any questions.

Subtract Fuel
--

Subtract Fuel
--

 

power_log_clear