jgyates/genmon #840: H-100 Controller Integration

hccunion, 2023-01-04:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

New setup to an existing H-100 Controller

### Actual Behavior

Not getting any data from controller - serial test is ok.

### Steps to Reproduce (including precondition)

Changed genmon.conf file for h_100 and modbuss address to "64" (100 from panel)

### Screenshot or Pictures relating to the problem (if possible)
[image]

h-100 screenshot of RS232 config

### Your Environment

- Generator Model: QTO7068GVSN
- Genmon Logs: Submit via the About page (submitted)
- Genmon Version: latest / new install
[image]
[image]
![PXL_20230103_221942408 MP](https://user-images.githu
[image]
busercontent.com/121971097/210627377-61e34d8d-d086-4b32-9a21-b32a426542a4.jpg)

hccunion, 2023-01-05:
I took this to one of our remote sites, and it came online and worked for about 30 minutes, but now it's not collecting data any longer. new logs submitted.

jgyates, 2023-01-05:
I received your logs. According to the data in your logs these are your communication stats:

Comm Stats : 
   Packet Count : M: 206, S: 0
   CRC Errors : 0 
   CRC Percent Errors : 0.00%
   Timeout Errors : 205
   Timeout Percent Errors : 99.51%
   Modbus Exceptions : 0
   Validation Errors : 0
   Sync Errors : 0
   Invalid Data : 0
   Discarded Bytes : 0
   Comm Restarts : 0
   Packets Per Second : 0.33

The means you sent 206 packets and have received zero. If it was working before then you must have restarted the genmon software. If loopback is working then either your controller is not receiving the data or it is not transmitting a response.

I am not sure from your previous post, has this worked before? or has the serial comms never worked? 

It it has worked, then I would validate both your Tx and Rx connection as well as your ground. A bad ground will not allow communication to occur.

If it has not worked then you may want to validate your pinout on the controller. Also possibly switch Tx and Rx and see if that makes a difference. DTE vs DCE pinouts can be confusing. For example one of them labels the transit as Rx as this is what it should connect to, not what it is.

Unfortunately I am not familiar with the H-100 setup in the controller so I can not offer any insight there.

hccunion, 2023-01-05:
Thank you! This may be a grounding issue - The H-100 is dumb in the fact you cannot have the serial cable hooked to the panel with the side cover on, there is no room. I had to take the DB9 out of the panel to plug it in. I will test it more tomorrow!

This was not working on generator 1, took it to generator 2, it was working, then stopped.

liltux, 2023-01-05:
@hccunion May I ask how you are powering the Raspberry Pi?

hccunion, 2023-01-05:
@liltux - currently the USB-C, swapping to the battery cable for the pintsize.me hat tomorrow. had the battery cable on generator 1, moved pi to generator 2 for testing with tbe usb-c ccable

hccunion, 2023-01-05:
@skipfire - using the DB9 hat. does that have the reverse polarity protection?

skipfire, 2023-01-05:
> @skipfire - using the DB9 hat. does that have the reverse polarity protection?

@hccunion right, that one does not, sorry.  Painkillers and tech-support apparently don't mix.

wb9rro, 2023-01-06:
I'm having a simular issue with my GenPi. Serial loopback test passes but I've got no communications with my H-100 controller. I have been rebuilding my genmon sice a failed software update. I went from 17.x to 18.16 (tried to) Failed.. I can now talk to genmon via ip:8000. I have setup the wifi using local keyboard, mouse and moinitor (raspi-config). I can get email notifications when I first power up or ask for a test message. Serial cable between controller and GenPi is good (Tested). Dose anyone know or have a dummy H-100 program that I could use to test with? My generator is not conveniently located and it's getting darn cold out these days. I've also struggled with getting the external WiFi (Alpha AWUS036ACS) to work. I made copies of the SD card after I got things working and stuffed it into another Pi 4. This second Pi 4 worked with the external USB Wifi until I did a sudo update. Back to zero.

liltux, 2023-01-06:
@wb9rro email me at liltux @ liltux .com. The drivers for the alpha adapter have to be loaded manually anytime you update the pi and the kernel is updated.  I can send you instructions.  As for the comms will need more info.  Email me and I can see what I can do to assist.  I was making those on eBay but is been awhile and a lot of other work in between.  So I am rusty.

wb9rro, 2023-01-06:
@hccunion  Thanks for detailing how you reset the H-100. How did you get to the serial communications menu on the display? I haven't paid much attention to the menu system on the H-100. Do you need to Generac commisioning software to open that up?

hccunion, 2023-01-06:
@wb9rro - no, you can navigate to that with the menu options - push "Menu", then navigate to "Diagnostics", and I think it's page 5 of the diagnostics for modbus slave (default option when you enter Diagnostics is "next page" (little arrow points to the right) - press the enter key to cycle through the pages.

wb9rro, 2023-01-06:
@hccunion  THANK YOU!!!

jgyates, 2023-01-07:
I am going to close this issue for now but feel free to post any follow up questions or status here.

closed 2023-01-07 by jgyates