jgyates/genmon #842: Data communication lost during outage

farberm, 2023-01-13:
My genmon setup works fine when there is no power outage. When the generator exercises it maintains data communication.

However as soon as there is a power loss, genmon reports data communication loss. 

If I restart the RPI via terminal command, I still receive a data communication loss error.

Only was to resolve the issue is unplug and then plug the RPI back in?

I send logs in during outage to see if you can determine what the issue may be???

jgyates, 2023-01-13:
your logs indicate that the serial port appears to be operating as expected, however they also indicate that packets are transmitted but nothing is returned. This points to an issue with your cable, possibly a grounding issue with your serial connection or the ground between the pi and the controller would be my guess.

Also, when you say "soon as there is a power loss", do you have any specific data regarding how long after a power loss? Often just the generator starting can cause vibrations that expose issues with intermittent cable connections. How are you powering your pi? If your using a plug in adapter with the pi then this could be the issue. Make sure you are power from either the battery or the controller.

jgyates, 2023-01-13:
How does the wall power supply maintain power during an outage when the generator is starting? Is it connected to a uninterruptible power supply (UPS)?

farberm, 2023-01-13:
I believe I have it plugged into a UPS but I will need to check to make sure tomorrow. However when the power comes back on it still does not receive data and I have to repower the unit.Sent from my iPad

farberm, 2023-01-15:
I checked and I had the RPi plugged into the UPS unit. I do have a ground wire running to the battery. Let me plug it into the wall and see if it corrects the issue. I would have thought that there was only a temporary data loss while the UPS provided power until the generator came on then the RPi would have the correct ground voltage. Only way to get it to work is disconnect from power source and restartThe RPi will lose power temporarily ( and reboot) when utility power is lost since I no no longer have it connected to the UPS. But it should restart without issue.Will this setup work or should I run a positive power wire yo the generator battery to the RPi?. This way it will not lose power and need to restartSent from my iPhone

jgyates, 2023-01-15:
The most uncommon method to power the pi with this project is using a wall power supply. Most people power from the battery or the controller.  Because these wall based supplies work on an AC to DC transformer, the DC output is not necessarily on the same ground as the AC supply. It is possible that the ground from the wall power supply transformer is close to the battery supply, but the UPS's battery ground is different and this is likely causing the issue you are seeing.

When you use the wall power supply the ground on the supply can change when using UPS vs utility power. 

You must make sure that the pi is on the same ground as controller when power is supplied by the utility and when running on generator power alone. If you power from the battery then you just need to make sure that the pi is on the common ground with the battery. Using the battery in the generator for power also alleviates the need for a UPS. 

You can try making the ground common between your house and the battery. In theory that **should** work, but I don't know any of the details of your facility. 

What I know will work is to power the pi from the battery and use the battery ground for the serial comms.

jgyates, 2023-01-15:
I am going to close this issue but feel free to post here for any follow up status or questions.

closed 2023-01-15 by jgyates