jgyates/genmon #843: Informational:  My build and setup journey

dmperl, 2023-01-14:
My build and setup journey:  https://imgur.com/a/96o8ZFy

### Your Environment

- Generator Model: {Please write here}
Controller : Evolution, Air Cooled
Rated kW: 16kW
Hardware Version : V1.06
Firmware Version : V1.21
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}
Generator Monitor Version : V1.18.16

JGYates / Genmon
https://github.com/jgyates/genmon
PintSize.Me Generator HAT
https://pintsize.me/store/ols/products/generator-hats-and-phat/v/PGI-GSTD
HighPi Case
https://www.amazon.com/dp/B07VNSNMVX
POE Splitter (microUSB)
https://www.amazon.com/dp/B01MDLUSE7
Copper Foil Tape
https://www.amazon.com/dp/B09Z6F9RFG
Faraday Cloth Tape
https://www.amazon.com/dp/B09LRRXB5V

jgyates, 2023-01-14:
Great info and pics. Thanks!

closed 2023-01-26 by jgyates

dmperl, 2023-01-27:
added Power Outage Log examples to https://imgur.com/a/96o8ZFy  (last page)
2 outage in 2 days. Genmon didn't miss a beat!