jgyates/genmon #845: My Genmon rebuild thinks it's talking to a Nexus controller. I have a H-100 controller

wb9rro, 2023-01-19:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

{Please write here}

### Steps to Reproduce (including precondition)

{Please write here}

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here}
- Genmon Logs: Submit via the About page
- Genmon Version: {Please write here}

jgyates, 2023-01-19:
This page has this info you are looking for:  https://github.com/jgyates/genmon/wiki/Appendix-G-Generac-H-100,-G-Panel-and-PowerZone-Controllers

I am guessing you have not selected the H-100 controller on the advanced page. Double click the gear icon in the top right of the web interface to access the advanced page. While you are there make sure you set the modbus address to the one your controller is using. That is usually 64.

closed 2023-01-19 by jgyates

liltux, 2023-01-19:
@wb9rro does the loopback test pass through the cable?  Cable could be damaged or bad ground reference.

wb9rro, 2023-01-19:
Cable is good. Used Serial tester at both endsSent via the Samsung Galaxy S22 5G, an AT&T 5G smartphone

jgyates, 2023-01-19:
do you have your modbus address set to 64 (100 is 64 hex) on the advanced page?

You will need to give more details on how your are monitoring your serial port with PUTTY? Is PUTTY running on the pi? If so is genmon running? If genmon is not running and your are getting data from the controller periodically then something is not configured correctly on the controller as the controller operates as a modbus slave, which means it will not transmit anything until it receives valid data from a modbus master.

wb9rro, 2023-01-20:
No I didn't have that set as I didn't know to do so. I am running Putty on 
a laptop with a null modem cable and serial tester on each end. My thinking 
is that genmon would be interrogating the generator controller on a 
occasional basis.  It appears that is true but the communications is not in 
ASCII. So I get strange characters.
I will try this weekend to make the changes you've recommended. I'll share 
with you all how I make out .   Thanks for your continued help.  Jim