jgyates/genmon #846: Stopping Generator during Power Outage

lakelivin, 2023-01-24:
I initially found the GenMon application searching for a way to cycle my generator during an extended power outage. I recently experienced a 21 hr utility outage. My generator ran continuously, and consumed 100 gallons of propane. My tank (250G) only holds 200 gal max. I am concerned in the event of a 3-5 day outage that I won't have enough propane.

I want to be able to switch to an intermittent generating cycle, where the generator runs for 20 minutes, then rests for 2 hours and continues to repeat this cycle indefinitely until utility power is restored. This will keep heat/AC inside the house comfortable and refig/frz cold, and reduce propane consumption to 12% of non-stop running (saving money and extending max run time).

I just got the GenMon running and was excited to be able to switch between Start/Transfer mode and Off. But this was with the utility power on. (Kudos also to @ripfire & Pintsize.me for the terrific Hat and Splitter! Made it so clean and simple.)

Further searching, after going all in on the GenMon, I found a recent Closed Issue "Stopping the generator from the web UI #821" where @jgyates advised "An automatic start can not be stopped by a remote command. This is a function of the controller, not the genmon software." I've also read multiple posts across the web about the advisability of removing the load before starting or stopping the generator, and performing a cooldown.

I've already implemented the Generac "2 wire start" modification in the J1/J2 connectors. In testing I found that will start and stop the generator when the controller is in Automatic, but won't work when the generator controller is in Stop mode. So when the N1/N2 sense wires have no power the generator is going to run in Automatic, and the physical 2 wire start switch will have no affect.

Whereas you already know so much about the controller and the transfer switch, can you advise a clean/reliable way to interrupt the Transfer control wire? I could use maybe an Arduino to control a couple of relays. One relay would close the 2 wire start switch and the other relay could switch the transfer switch back to Utility to remove the load from the generator. The Arduino could handle the timers for on/off, load removal, and cooldown. But I would like to benefit from your much greater knowledge and experience.

- Generator Model: Guardian 22kw Air cooled / G70438 / EVO 2 controller / 11/2021 manuf date
- ATS: 200A SE  (Model RXSW200A3)
- Genmon Version: V1.18.16

jgyates, 2023-01-24:
@lakelivin 

That is an excellent question. Unfortunately I don't know enough about the transfer switch to answer that question with confidence. there are others that use the software and monitor the forum that would likely be more help than I could.

You could try the software remote commands and see if they work. I was told by a user that they did not work although I did not validate that myself with an outage at the transfer switch.

I don't think the transfer switch will switch back if it does not sense a utility voltage as this is a safety feature for the people working on the lines. They do make load shed devices that allow selective circuits to be removed from the generator load, although this requires wiring changes to be made in each circuit. My transfer switch has something like this on it: 

https://www.amazon.com/Generac-Load-Shed-0K0220ASRV-0K7341A/dp/B016AQ8WLQ

However I do not use this feature so I can't really advise a lot about it other than it is designed to remove individual circuits from the generator load to save fuel during an outage.

skipfire, 2023-01-24:
Does 2-wire work in manual mode?

lakelivin, 2023-01-26:
Thanks @lmamakos, that makes sense.

grsthegreat, 2023-01-30:
did you change the software inside the evolution controller to TELL controller that your using 2 woire start? if you change that setting, it will disable the auto mode. this will fix your issues.

lakelivin, 2023-01-31:
Hey @grsthegreat, fancy meeting you here!

When I received the genuine Generac 2 wire start kit it contained instructions on how to install the connectors and how to change the Dealer configuration setting for 2 wire start. The directions included the key sequence to access dealer level settings. 

I followed the directions exactly and changed the setting, then backed out, set the generator to Auto and confirmed operation of the 2 wire start.   (Directions copied below.)  But I found afterwards that if I turned off the Main house power switch in the ATS that the generator would start in Auto, and that then the 2 wire start switch had no affect.  If I left the Main on, and the generator in Auto, I could use the 2 wire start switch to start the generator and to stop it, only after starting in 2 wire mode.

Use the up/down arrow keys to navigate to the SUB MENU field and then press ENTER to display the sub menu.
5. Use the up/down arrow keys to navigate to the DEALER field and then press ENTER to display the dealer menu.
6. Use the up/down arrow keys to navigate to the DEALER-EDIT field and then press ENTER to display the dealer
edit menu.
7. Use the up/down arrow keys to scroll through the parameters until 2 WIRE START SELECT is displayed. Press
ENTER to select this parameter as the one to be edited.
8. Use the up/down arrow keys to change the display to read YES.
9. Press ENTER to store the data. Remote start is now enabled.
10. Turn utility MLCB to ON or Closed.
11. Place the generator switch in the AUTO mode (green light on the control panel is ON).
12. After completing installation, verify that the two wire start is functioning properly.

grsthegreat, 2023-01-31:
Thats off, switching it into 2 wire mode shuts off the necessity ti even run 240v to the generator. It should not start. Go into dealer menu and make sure its in 2 wire mode.I have quite a few off grid units set to 2 wire start and they dont have 240 run to it, and they work just fine.Sent from my iPad

lakelivin, 2023-01-31:
This is what I think that I am going to do:

I have a "Cyclic timer relay Selec 55XC-P8-CU" 
[image]
[image]
[image]
This is an electronic timer relay with 2 NC and 2 NO contacts. The relay has selectable on duration and off duration. Power for the timer and the relay closure can come from just about anything including 120V AC, 240V AC, or 12V DC. The switch is rated for contact current of 5 amps at 240V, so it is plenty safe enough.

I'm going to wire pins 1 & 8 to the N1 N2 sense connectors on the Evolution controller.
I'll wire pins 4 & 5 to the 240V N1 N2 sense conductors coming over from the ATS. If the relay is "Off", or if it has no power to it all, the N1 N2 conductors from the ATS will pass directly through to the Evolution controller.
I'll wire pins 3 & 6 to a 12V dc to 240V ac inverter connected to the generator battery.

A toggle switch will switch 12V power from the battery to both the inverter and the timer pins 2 & 7. So the relay and its electronic timer will run off 12v DC, but the contacts will be switching 240V AC. Both the inverter and the timer will be dead unless I switch on the toggle switch. I will only do this when it becomes apparent that I'm going to have a long Utility power outage. Initiating this modification will not be automatic. I'll have to go outside and turn it on once. Once it is turned on, it will continue to repeatedly cycle on the timing I select.

When I want the generator to Stop, the relay will send invertor 240V power to the Evolution N1 N2 sense terminals. (This will open the connection to the N1 N2 sense conductors coming from the ATS, so no power will backfeed into the Utility. The Evolution controller will think that Utility power is restored and will automatically signal the transfer switch to switch back to Utility, run the 1 minute cooldown cycle on the generator engine, and shut it down.

I am going to have to use a relay in the ATS also to switch the power to the coil for closing the Transfer Switch to Utility. Doing this will remove the load from the generator. I also will not send any generator power out through the Utility connection because the transfer switch is going to open from the Generator side.

When I want the Generator to restart, the relay will open from the 240V invertor and close to the dead Utility N1 N2 wires. The ATS is already switched to Utility, so there is no load applied to the Generator. The Evolution controller will start the generator engine, perform a warm-up, then will command the ATS to switch to Generator.

I'll come back in a few days with some more thoughts on control of the additional relay in the ATS. I have a few other things that I am working on right now and won't get back to it for a couple of days.

I'll be waiting to see what y'all think about this plan. It is quite simple, I think. It is one cyclic timer relay, on inverter, and another relay in the ATS.

lakelivin, 2023-01-31:
Hey grsthegreat, you're much quicker response came in while I was typing out my long response.

I'm going to have to go out and play some more with the 2 wire start and the Main power switch and make sure I'm correctly understanding what happens - before I start messing with this other stuff.

Thanks for the info. I'll go back and check out what you said.

jgyates, 2023-02-01:
I am going to close this thread but feel free to post any updates here.

closed 2023-02-01 by jgyates