jgyates/genmon #847: "Mem checksum failed" error when GenMon connected to Nexus controller

nitroxjunkie, 2023-01-25:
I have what seems to be a unique issue that searching the web hasn't come up with anything helpful.  

On my Pi I got GenMon loaded up just fine and I am able to access the interface.
On my generator, I pulled the controller fuse, connected the Pi/hat, inserted the controller fuse and immediately got a "mem checksum failed" error on the controller.  so I pulled the fuse, unplugged the pi from the controller, put the fuse back and the controller booted just fine.  Hoping it was a one-off issue I again pulled the fuse, connected the pi to the controller, inserted the fuse and immediately got the same error.  

With the pi disconnected from the controller, the generator is fine and runs happily.  With the pi connected the controller goes belly up.

now I am at a loss, I cant figure how the pi would cause a controller memory corruption - only while connected- or what could cause this issue.  Has anyone experienced anything similar or can offer any troubleshooting advice?

nitroxjunkie, 2023-01-25:
Thanks for the near immediate response :)

1. The pi is a 4B 2Gb
2. power is from the battery using the included cable from pintsize.me through the barrel connector, not from the molex.  The dip-switches are set as 
GENPWR: OFF(1)
BUCK: ON
3.  I'll try to get some pics tomorrow if the snow allows.  It's connected with the 8-pin molex cable from pintsize.me so there isn't much to mess up.  (famous last words).  The only 2 wired connections on the hat are the molex and power barrel - nothing else is plugged in anywhere on the pi or hat.

jgyates, 2023-01-30:
I am going to close this issue but feel free to post and follow up questions or status here.

closed 2023-01-30 by jgyates

nitroxjunkie, 2023-06-04:
I finally got time to circle back to this project.  I went at it with a brand new Pi 4b 4Gb, brand new SD card with a fresh install of GenMon.  I got everything assembled and had the same results - when I boot the Nexus controller, with the GenMon hat attached I get the same "Memory Checksum Error." I did test with only the Molex cable connected to the controller (pi not connected) and I didn't get an error.  If I connected the Pi Hat to the Molex cable then the error occurred.  Pics attached - Please excuse that I didn't have the Pi in a case - I didnt want to invest in parts for a case and mounting until I knew it would work.

Thanks!

[image]
[image]
![Image (2)](https://githu
[image]
b.com/jgyates/genmon/assets/63306111/70c17d98-2b20-471f-aa2f-3ea6c7f6cb01)

jgyates, 2023-06-04:
This thread has a similar issue (blank screen instead of mem checksum error) with Nexus controllers that turned out to power related if memory serves:

https://github.com/jgyates/genmon/issues/834

I recall a few issues with some Nexus controllers but they were related to the pi w, not pi 4. It could be pintsize.me issue. @skipfire may have some suggestions as well.

skipfire, 2023-06-04:
The serial circuit on the HAT has not changed in any meaningful way since the first iteration and I know that others have gotten it working on the v2.1 HAT, but I know at least one user gave up on trying to get it to work with Nexus (hence the warning now on the product pages about questionable compatibility with Nexus).  

I would suggest progressive troubleshooting.  Stop at the first one of these to generate the Checksum.
1. Molex plugged into HAT without Pi
2. Molex and power plugged into HAT without Pi
3. Molex plugged into HAT & Pi, but no power connected to the HAT & Pi (the Pi should not be powered here)
4. Plug power into the HAT & Pi, stop GenMon, then plug in the Molex cable
5. With everything connected and GenMon stopped, start GenMon

This series of tests narrows down potential hardware related causes one at a time, though recently someone had a controller that required new settings to work so it's possible that something like that is in play here as well.  You may want to go through the Settings and Advanced Settings in GenMon to see if anything looks funky, since it seems to not be getting initial communications it probably has not detected the controller type to do any related configurations there.

skipfire, 2023-06-05:
I'm surprised that it threw the Memory Checksum Error when connected to the HAT without Pi or power connected, that means that being connected to an unpowered TTL chip is enough to cause the problem, which I find completely mind boggling and leaves me without any other ideas to resolve the issue.  If you use the contact form on the PintSize.Me page or reply to the order confirmation email (if you still have it), I can arrange for you to return the purchase and get you a full refund.

nitroxjunkie, 2023-06-05:
Thanks Jon - I know the chances are very slim, but whats the likelihood
that it could be a bad hat or TTL chip on the hat that it would be worth
trying a different hat?  Or is it most likely that my controller is just
old and cranky so it's refusing to cooperate?

Thanks!

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality11&>
06/05/23,
01:55:40 PM

skipfire, 2023-06-05:
I test every unit prior to shipping so I know the board worked then.  It is possible that something failed after that but pretty unlikely.  If I get it back I will examine it both under my electronics microscope and on my generator to confirm that the board is good.  Based on what I've heard from a few other Nexus owners I'm leaning towards the cranky option, but when I get the board back if I find any problems with it I can ship a replacement instead of refunding if that's what you want, but I'd suggest we take further discussions through direct email.

lmamakos, 2023-06-05:
Maybe try turning off the GENPWR switch so you're not backfeeding power
from the Raspberry Pi into the generator controller?  That's my guess from
looking at your photo and seeing the DC power plug connected?

Dsunland, 2024-08-17:
@nitroxjunkie did you happen to fix your issue?  I am having the same issue.

Thank you!