jgyates/genmon #848: Basic authorization header support?

skipfire, 2023-01-28:
Is there any support for basic authorization?  I'm wanting to setup an e-ink display running on an ESP32 (specifically https://www.lilygo.cc/products/t5-4-7-inch-e-paper), but MicroPython has only a very limited requests library that doesn't support cookies or redirect.

jgyates, 2023-01-30:
No, that has never come up as a requested feature. If you can give me more details I can see if that is something that can be easily implimented.

skipfire, 2023-01-30:
For basic auth you can pass a header value on an API call.  The key is `Authorization` and the value is `Basic {authstring}` where authstring is a base64 encoded `{username}:{password}`.

jgyates, 2023-01-30:
I assume you tried the approach here and it did not work with MicroPython requests library?

https://github.com/jgyates/genmon/wiki/Appendix-C--Interfacing-Generator-Monitor-to-External-Applications#web-api

If you just want to get the data you can use the socket interface. Only the web interface uses the password. The socket interface that is used with ClientInterface.py is not protected by passwords and is on a different port number. In short, genmon does not have a password for getting the data. genserv.py, the web interface, does protect access with a password. So you can use the ESP32 with a socket interface to get the data and display it using sockets with MicroPython.

skipfire, 2023-01-30:
MicroPython's implementation doesn't support session, cookies, or the redirect that happens after a successful login post so it throws an exception.  Right now I've written a proxy in full python that runs on port 8001 that accepts the basic auth, does the stuff you mentioned, and returns the payload for use on the ESP32 running MicroPython.  As for ClientInterface.py I'll probably look at options to block remote access to that port since it isn't secure.

closed 2023-02-01 by jgyates