jgyates/genmon #850: Install on Ubuntu

LPCALC, 2023-01-30:
Try to get Genmon up & running on a VM running Ubuntu 20.04.3 LTS. Install seamed to go off without a problem. 

However, when starting, Receiving error: 

~/genmon$ ./startgenmon.sh start
_Starting genmon python scripts
Warning: required library spidev not installed. Attempting to install....
Error in InstallLibrary using pip : spidev : UnInstall: False: b'  ERROR: Command errored out with exit status 1:......._

once passed the error, genmon appears to start

_Running one time maintenance check
Starting....
Starting /home/lpc/genmon/genmon.py 
Starting /home/lpc/genmon/genserv.py_ 

Unfortunately I am not able to access web GUI to complete the  Serial over IP setup.

jgyates, 2023-01-30:
Give it a minute or two, you are probably getting a lot of serial timeout errors and the GUI will not load until the init sequence has completed so you have a few minutes of timeouts. Alternatively you could edit the /etc/genmon/genmon.conf file directly.

https://github.com/jgyates/genmon/wiki/1----Software-Overview#configuring-genmonpy

It may be waiting on serial port to become active. I remember puting in some code to wait on the serial port (for people using USB serial). If that is the case the you will need to edit the /etc/genmon/genmon.conf file directly.

also the spidev is for the SPI port on the raspberry pi. Not a show stopper for using in a VM, just limits the use of some add ons.

LPCALC, 2023-01-30:
Thanks for the quick response.   I waited a while, but did not get access to GUI.  I edited the genmon.conf file manually. When I restarted _(startgenmon.sh restart)_  it indicates that the processes where not started to begin with. 

_Stopping....
Stopping genserv.py
kill: (14387): No such process
Stopping genmon.py
kill: (14382): No such process
Starting....
Starting /home/lpc/genmon/genmon.py 
Starting /home/lpc/genmon/genserv.py_ 

Does genmon & genserv terminate after a period of time when not serial connection is detected?  If not my install may have not gone as I expected.

jgyates, 2023-01-30:
If there are no serial port present, then yes, genmon will exit. You can override this behavior by setting forceserialuse=True in /etc/genmon/genmon.conf

genserv will exit of genmon is not loaded.

capindc, 2023-05-03:
Hello jgyates this is the closest thing I have I am running ubuntu in a VM sorry kind of new so sorry if I do not get things right away this is a new project for me.
here goes 
sudo python3 genmon.py
I get 
Error on open serial port /dev/serial0: [Errno 2] could not open port /dev/serial0: [Errno 2] No such file or directory: '/dev/serial0'
Error on serial port init!
Error opening modbus device

can you help

jgyates, 2023-05-04:
This error is due to either a serial port not being present or mapped for use in the VM or the name of the serial port is not /dev/serial0. The default serial name in genmon is /dev/serial0 as this is what it is on a the Raspberry Pi OS, which is a debian variant.

If the genmon is setup to use a physical serial port and there are no serial port present, then genmon will exit. You can override this behavior by setting forceserialuse=True in /etc/genmon/genmon.conf, however you either need a serial port present or you need to use a serial over IP device and tell genmon to use the network instead of the serial port.

Typically if genmon is used in a VM it would be setup for a serial over IP.  You can make this serial over IP settings change on the settings page of the web interface. If the web interface is not active (because genmon has exited due to a serial port not present) then you can edit the file /etc/genmon/genmon.conf 

Here is the default file:

https://github.com/jgyates/genmon/blob/42613950d0b0979db0e9d036befd62437f58d8a0/conf/genmon.conf#L8

then change 

use_serial_tcp = True
 
and then set serial_tcp_port  and serial_tcp_address according to your setup.

jgyates, 2023-05-04:
Also, to go further you need to pick a method of how you want to use genmon (direct serial connection or serial over IP). For direct serial port, you need a serial port on the system (or accessable to the VM). For serial over IP, [this article has more info](https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP).

capindc, 2023-05-04:
If I change it to tcp will I be able to access GUI

capindc, 2023-05-04:
Thanks but I am still getting the same problem this is my genmon.conf file. I have not set the TCP address yet becouse I have not gotten it yet.
[GenMon]

# name of the site, used in the web interface and email output (required)
sitename = SiteName
forceserialuse = True

# the serial device name of your serial port. Normally  /dev/serial0 or
# /dev/ttyAMA0 for onboard Raspberry Pi Serial ports (required)
port = /dev/serial0

# Set to true to enable Serial over TCP. A hardware serial to TCP converter is
# required and must be setup for 9600,N,8,1 for most controllers. The software
# will only make a TCP socket connection, it will not setup the port. The
# software expects the TCP connection to be a passthrough (i.e. not process
# modbus protocol). The project software will process the modbus protocol.
# If modbus_tcp is enabled the instead of serial passthrough over TCP, the
# modbus TCP protocol will be used.
use_serial_tcp = True

# if use_serial_tcp is enabled, this parameter sets the IP address used.
serial_tcp_address =

capindc, 2023-05-04:
can I put in any ip adress and port number I have not gotten my USR-W630 yet

jgyates, 2023-05-04:
I would put an unused IP address on your network. and use a port number above 1000 to avoid conflict wtth any other network devices.

capindc, 2023-05-04:
I should be just be able to start genmon.py on it's own right with out the Serial port error.
Sudo Python3 genmon.py 
no matter what I do i get the same error

capindc, 2023-05-08:
Find Answer below
Did you run the set instructions here: https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software...yes

Were are you seeing this error (on the console, in a log file, if so what log file)? Genmon uses several log file in the /log folder (genmon.log, myserial.log, myserialtcp.log are some of the relevant ones, depending on your settings)...NM Iwas reading it wrong

What is the timestamp on the error and you you seeing a new error each time you start are you referencing the same error. Genmon does not delete the log file each time it is started.
where is the location of the file named genmon.conf you modified? genmon/conf
you can attach files to this thread via the github web interface. the files /log/genmon.log, /log/myserial.log and /log/myserialtcp.log would be relevant.  ...yes here are the file the myserialtcp.log is missing

[genmon.log](https://github.com/jgyates/genmon/files/11421105/genmon.log)
[myserial.log](https://github.com/jgyates/genmon/files/11421106/myserial.log)

jgyates, 2023-05-08:
Modifying the genmon/conf file is your problem. The files in ./genmon/conf are the original unaltered conf files. the working files are in /etc/genmon/*. As I mentioned a few times in this thread, you can modify the file /etc/genmon/genmon.conf to change the settings if you are unable to load genmon due to the absence of a physical serial port. The fact that you do not have the file /var/log/myserialtcp.log indicates that you have never enabled serial over TCP.

The reason for this file location is that if you were to modify the file in ./genmon/conf then the file would be overwritten when you updated the software from the github repository, hence the working copy is in the /etc/genmon folder.

capindc, 2023-05-09:
OMG I got thank you sir