jgyates/genmon #851: Fuel consumption and Run hours show 0.00

lmackley, 2023-01-31:
On maintenance page fuel and run show 0.00.
Generator Monitor Version : V1.18.16
I've included part of the maintenance page as well as part of my run log.
If you need any other info, let me nw what you need.

Maintenance :
    Model : G0062620
    Generator Serial Number : 0008632332
    Controller Detected : Evolution, Air Cooled
    Nominal RPM : 3600
    Rated kW : 20 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Natural Gas
    Generator Phase : 1
    Engine Displacement : 999 cc
    kW Hours in last 30 days : 0.00 kWh
    Fuel Consumption in last 30 days : 0.00 cubic feet
    Total Power Log Fuel Consumption : 0.00 cubic feet
    Run Hours in last 30 days : 0.00 h
    Run Hours in the last year : 0.00 h

    Outage Log : 
        2023-01-19 13:31:24, Duration: 0:00:11, Estimated Fuel: 0.00 cubic feet
        2022-11-30 11:37:07, Duration: 0:00:10, Estimated Fuel: 0.00 cubic feet
        2022-11-30 11:23:57, Duration: 0:00:10, Estimated Fuel: 10.55 cubic feet
        2022-10-24 14:24:25, Duration: 0:00:11, Estimated Fuel: 0.00 cubic feet
        2022-10-14 08:57:35, Duration: 0:00:11, Estimated Fuel: 0.00 cubic feet
        2022-10-06 12:56:40, Duration: 1:37:27, Estimated Fuel: 202.64 cubic feet
        2022-09-28 16:52:53, Duration: 0:28:50, Estimated Fuel: 182.18 cubic feet
        2022-09-28 09:47:37, Duration: 0:57:21, Estimated Fuel: 105.16 cubic feet
        2022-09-28 08:50:22, Duration: 0:00:11, Estimated Fuel: 0.00 cubic feet
        2022-08-30 11:43:16, Duration: 0:00:11, Estimated Fuel: 0.00 cubic feet

Thanks

jgyates, 2023-01-31:
I am not sure if you are saying that you had an outage of 11 seconds or that there is an outage of 11 seconds reported but you did not experience that. Logs are always helpful to put things in context.

For an outage of 11 seconds I would not expect any fuel to be reported used for 11 seconds as the calculation used is not that accurate to detect any fuel during this time. Also this includes the 5 second delay the controller uses before starting the engine.

If you are getting a false outages of very small duration then I would look at communication errors first. Are you using serial over TCP/IP? If so dropped packets over wifi can cause timeouts and a late reply from the network due to wifi network retries. See number 16 on Known Issues (https://github.com/jgyates/genmon/wiki/Appendix-D-Known-Issues)

Also, this issue has some details on two advanced settings you can change to delay the reporting of outages: https://github.com/jgyates/genmon/issues/742

I am going to close this as I suspect this is a communication issues. Logs would validate that, but you can look at your serial stats on the monitor page and see if you have any timeouts. Note that serial stats reset if genmon is restarted so this would only be an indicator if genmon has not been restarted since your reported outage. If you still have issues or questions on this topic post them to this thread.

closed 2023-01-31 by jgyates

lmackley, 2023-02-01:
There are the lines I was referring on the **maintenance** page.

Fuel Consumption in last 30 days : 0.00 cubic feet
Total Power Log Fuel Consumption : 0.00 cubic feet
Run Hours in last 30 days : 0.00 h
Run Hours in the last year : 0.00 h

I understand the last 30 days may be 0 but the others I believe should have a value other than 0.

jgyates, 2023-02-01:
I would need to see your logs and your power log (/etc/genmon/kwlog.txt) to give any meaningful insight.

These are the commands you can run form the browser to get the same data assuning and IP address of 192.168.1.10 and http(not https):

Fuel in 30 days
http://192.168.1.10:8000/cmd/power_log_json?power_log_json=43200,fuel
Fuel all time:
http://192.168.1.10:8000/cmd/power_log_json?power_log_json=0,fuel
Run hours 30 days
http://192.168.1.10:8000/cmd/power_log_json?power_log_json=43200,time
Run hours last year
http://192.168.1.10:8000/cmd/power_log_json?power_log_json=525600,time

griffths, 2023-06-30:
I am having the same? problem with the fuel consumption estimate.

I understand that the amount used for a short (less than 1 minute?) run would be difficult to calculate but I recently had an outage of approximately 2 hours and the estimate still displays zero CF.

I searched and found issue 851 and gathered the stats from the logs but all were zero.  I suspect that this is due to the fact that the Pi is disconnected from the controller at the moment while I perform some maintenance/upgrades to it.

I did attach screenshots of the outage and settings page but can provide the other four stats once I get the Pi reconnected to the controller if needed.

[image]

[image]

regards,

Jay

jgyates, 2023-06-30:
I can't say for sure I I don't have your logs, but based on the setting above "Use Nexus Legacy Frequency" it looks like you may have a Nexus controller, which does not support a reliable way to determine power. Unless you have an external set of current transformers then this function would not be supported by your controller. Knowing the fuel consumption rate, which you put in the above settings is needed but calculate the fuel used, however you also need to know the power used by the generator at a given time. [This page has more info](https://github.com/jgyates/genmon/wiki/Appendix-H-Monitoring-Fuel-and-Power-Usage).

griffths, 2023-06-30:
Thanks for the quick reply.

Your explanation makes perfect sense.  No way for the software to know the load on the Genset without the sensors which I understand are not part of my controller from Generac.

I am in the process of adding the current transformers using the Sensor HAT from Pintsize.Me