jgyates/genmon #852: Raspberry Pi Shortage

jmcjclark, 2023-02-01:
I have been using this project for years now and absolutely love it! I am trying to set one up for my dad and Raspberry pi’s apparently are in short supply these days. Is this program compatible with any other raspberry pi alternative such as orange pi, banana pi, potato pi? Thank you!

skipfire, 2023-02-01:
Mine is running on a Nano Pi Neo Air-LTS that I picked up on Amazon, though I do have a custom HAT for it as well.  There is a wiki page for non-Raspberry Pis that started from me documenting setting it up on a couple of non-Raspberrys.  There may be easier paths than what I got working, but I was able to get it working.  Watch for WiFi as not all of the alternatives have it onboard.  https://github.com/jgyates/genmon/wiki/Appendix-Q---Using-Non-Raspberry-Pi-Hardware

jmcjclark, 2023-02-01:
Thanks for the quick reply! I have purchased one of the hats on pintsize.me before. Is your nano pi neo  air-lts hat available?

skipfire, 2023-02-01:
Hit me up through the PintSize.Me contact page and I can get you some details on it, it is quite different and only 5 exist right now.  The Nano Pi Neo Air was a bit difficult to get WiFi working on, the Nano Pi Neo Plus 2 was much easier to setup since it has an ethernet port.

jgyates, 2023-02-01:
Also, If you use serial over IP genmon will work on just about any linux platform. The only thing that is specific to the pi is the docs and utilities setup the pi serial port with the install script. The UI does display a couple of pi specific things like under voltage detection on the pi, but those are for information only and not required for operation. Other than that the software assumes a linux file system and network stack.

closed 2023-02-01 by jgyates