jgyates/genmon #853: having trouble seeing my generator on computer, but get emails and pushes

grsthegreat, 2023-02-01:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

{Please write here}

### Actual Behavior

i had to reboot my wifi router, and now i no longer can log into Genmon

I went ahead and rebooted the Pi unit, and logged it onto my network. i get all the pushes and it even gives the same ip address as i see the generator on Fing, but when i type that address:8000 into my computer, it says unable to connect. i have tried rebooting the pi unit 3 times, and if i do a manual gen shutdown i get the push notification, but cant log into the unit.

not sure whats wrong. my wifi log shows generator is actively logged onto wifi system.

jgyates, 2023-02-02:
When genmon restarts it sends and email that notifies genmon is starting. This email contains the IP address of the pi that genmon uses. Validate that this is the same IP as Fing. Also, if the ip address is the same then you should be able to ssh into the pi to log in at the console over the network.

Alternatively you can access gennon with the [ClientInterface.py program](https://github.com/jgyates/genmon/wiki/1----Software-Overview#clientinterfacepy).

Since genmon appears to be running my best guess is that, depending on your router model and feature, your router may be blocking webservers that are local on your network. Some of these issues in the past have been marked with the "Router" tag so you can view them here: https://github.com/jgyates/genmon/issues?q=label%3ARouter+

grsthegreat, 2023-02-02:
Well, im at a loss. The email ip address is the same as fing shows. I log into router and the ip address shows its active and allowed. Still nothing. 

I tried port forwarding port 8000, still nothing.

grsthegreat, 2023-02-02:
i can log into putty using ip address.

jgyates, 2023-02-02:
Are you using remote access and exposing your pi to the internet? Is your router an ASUS model?

jgyates, 2023-02-02:
If you have an ASUS router then read this thread: https://github.com/jgyates/genmon/issues/271

grsthegreat, 2023-02-02:
I am not exposing it to the internet, i only tried port forwarding as a possible issue. I have a netgear router

jgyates, 2023-02-02:
Check your router for any security related settings that may block local servers on you local network. The ASUS feature is called AI Protect and it decides what to block on it's own. If you have a feature like that then try disabling that feature.

skipfire, 2023-02-02:
I've got 3 suggestions.

First, make sure that the port is listening on the Pi.  `sudo netstat -tulpn | grep LISTEN` will return a bunch of rows, you are looking for one that has 0.0.0.0:8000 in the third column.  If you see that entry, then GenMon is listening, if you don't, GenMon probably isn't listening.

Second, make sure your browser isn't swapping it to https, several browsers have started automatically replacing http with https even if the site can't handle it.  It is supposed to not do it for local stuff, but I don't think it always guesses what is local correctly as I've had that issue a couple times.

Third, try using telnet on some device that isn't the Pi to see if it can connect to port 8000.  On windows (telnet needs to be added through Add/Remove as it is not installed by default for some dumb reason) it would be `telnet ipaddress 8000`.  If it works it will just give you a blank screen, maybe a blinking prompt, and the Windows telnet client exits with Ctrl+].  If that works then it shouldn't be a network issue.

skipfire, 2023-02-02:
Might take a look at the genmon logs in /var/log/ then, maybe it's logging an error about why it can't start.

jgyates, 2023-02-02:
@grsthegreat 

Can you post the output of your putty session? The information your are providing is not painting a clear picture. You initially stated that the problem started when you rebooted your router, which implies the issues is with your router. Now you are implying that that port 8000 is not listening, which implies an issue with your pi. Please post the output of the test that shows port 8000 is not listening and did you run this test on the pi or some other computer?

grsthegreat, 2023-02-02:
[image]

grsthegreat, 2023-02-02:
i assumed issue was with router as this happened after my reboot. now im not so sure

jgyates, 2023-02-02:
Did you wait at least 30 seconds after rebooting the pi before trying this test? The cron job that starts genmon has a 30 second delay before genmon is started. This delay gives the linux network subsystem time to fully start.

jgyates, 2023-02-02:
you can type this command to determine how long the system has been up:

    uptime

Wait until uptime shows at least one minute, then run netstat. If you do not see port 8000 then restart genmon from the command line and show the output here:

      cd genmon
      ./startgenmon.sh restart

grsthegreat, 2023-02-02:
just tested it again, uptime 53 minutes

same screen, dont see 8000

jgyates, 2023-02-02:
submit your logs on the about page. I will see if there are any obvious errors in the logs.

grsthegreat, 2023-02-02:
just submitted logs

grsthegreat, 2023-02-02:
im still not sure why rebooting router would cause this issue, but i want to thank all of you for assisting me with this issue

grsthegreat, 2023-02-02:
also, i just sponsored this site for helping keep this wonderful program alive

jgyates, 2023-02-02:
I looked at your logs, no recent errors and nothing looks out of the ordinary. Let me know if it occurs again.

jgyates, 2023-02-02:
I am going to close this for now.

closed 2023-02-02 by jgyates