jgyates/genmon #854: Powering pi

aric8456, 2023-02-05:
I have a pi 4 and used this https://smile.amazon.com/HOMREE-Step-Down-Converter-Module-Charger/dp/B07ZQB6S3L/ref=sr_1_5?dchild=1&keywords=12v+to+5v+dc+converter+3a+usbc&qid=1601328672&sr=8-5  and bought small alligator clips that I wire nutted and tried to affix to the battery. It kinda worked but now there's corrosion on one of the battery terminals and it doesn't work anymore.

What is the best way to attach that transformer to the battery?

skipfire, 2023-02-06:
A motorcycle USB adapter is a good choice, here are 2.  Just watch the amps/watts they can put out.  3A/15W is ideal, the first link below is listed for 2.4A/12W.  Don't go below 2.0A/10W with a Pi 4.

https://www.amazon.com/Suuwer-Motorcycle-Charger-Charging-Smartphone/dp/B07WXKPTS3

Another option is this one, while I'm not sure I'd want one that has an on/off switch, it may give you ideas of similar products you can look for.

https://www.amazon.com/65W-12V-USB-Outlet-Laptop/dp/B09YTWNCM2

To install these I use a vice grip to hold the terminal cover closed tight, remove/loosen the nut that secures it, place the loop or fork in place, tighten nut, remove vice grip and repeat on the other side.  When doing the positive terminal make sure your metal tools do NOT touch any part of the generator or you will short the positive to ground.  I always just put one hand against the metal that the wrench could touch so that it smacks my hand and not the metal.

jgyates, 2023-02-06:
I updated the wiki with the link above.

closed 2023-02-06 by jgyates