jgyates/genmon #855: Utility Voltage at 84v and Auto Starts When Pi Plugged In

GitMoGen, 2023-02-06:
Hello Team,
I plugged in my Raspberry Pi 3 with the PintSize.me top hat to the generator.  All software loaded and the pi does communicated with my 10kw Evolution 2 controller running V1.11 of firmware.  The generator is located 75ft ware from the transfer switch.  I had the generator turned to off during the connect of the Molex plug.  I am using the Molex cable harness to power the pi.  I am NOT using the battery supplied power cable (came with the pi hat) or USB power at this time, just the Molex Cable to power the pi.

I noticed immediately on the GenMon web interface that the Utility Line Voltage was at 84 volts.  When I switched the generator to Auto, it believed that the Utility Voltage was Low (84v) and auto started the generator.  I have confirmed that the true utility voltage is normal at the transfer switch.

My question is, it seems that the pi is drawing down the utility voltage feed from the control panel and thus thinking the utility voltage is low.  Am I on the right track here or is something else going on?

Should I switch power to the Pi over to the battery harness or USB to see if the issue goes away?  

Any help would be great!

BTW - Excellent software and pi hat

Thanks!

Mo

liltux, 2023-02-06:
@GitMoGen 1st, I would not power a Rpi 3 from the board.  The dc output of the molex is limited and not rated to power a Rpi 3 or 4.  So yes you would want to get battery power for Rpi 3 board. 2nd You need to check the utility sensing power at the generator and not the transfer switch.  There should be a Wago block labeled N1 N2 T1, you need to check for 240 VAC at N1 N2 at the Generator.

GitMoGen, 2023-02-06:
Thanks for your quick response, I will give it a try!

Thanks,
Mo

GitMoGen, 2023-02-06:
Thank you, will be giving it a try!

skipfire, 2023-02-06:
@GitMoGen I can confirm that what you are seeing is normal when there is too much power being drawn from the controller.  It seems to throw off the voltage sensors inside the controller making it think that the utility power is much lower than it actually is.  Using the power cable into the HAT should solve that issue.

GitMoGen, 2023-02-06:
Problem solved, thanks for the quick response.