jgyates/genmon #856: Temperature gauge display problem

T-C123, 2023-02-10:
**Expected Behavior:**
Temperature gauge displaying analog and digital temperature

**Actual Behavior:**
Displays temperature but the digital display for the second temperature gauge is displayed low, almost off the bottom of the gauge.

**Steps to Reproduce** (including precondition):
Whenever Status page is displayed

**Screenshot** or Pictures relating to the problem:

[image]

**Your Environment:**
Display is the same on Chrome, Firefox or via Real VNC.

**NOTE:** The gauge displaying incorrectly is the second of two added under Add-Ons and data is retrieved from a thermocouple.

[image]

- **Generator Model:** Generac 18 kw, Evo2
- **Genmon Version:** V1.18.16

This is a minor problem but I thought you should know.

Thanks for your help!

jgyates, 2023-02-10:
Try shortening the label form "Engine Oil Temp" to something like "Oil Temp" or "Oil Temperature". The issues is that all of the gauge placements expect a single line for the label. The gauges use a 3rd party library so I don't have full control over the spacing. Let me know if that does not fix your issues.

closed 2023-02-10 by jgyates

T-C123, 2023-02-10:
Fixed, Thank you!