jgyates/genmon #857: Genmon not updatig

wrvolz, 2023-02-10:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

If you are having other issues, please see the following page:

https://github.com/jgyates/genmon/wiki/3.5---General-Troubleshooting

You can search past issues by removing the "is:open" keywords from the search window and adding any keywords that are relevant to your issue.

If you need to send you logs and registers to the developer, if your email is setup and working properly you can click send your logs on the About page in the web interface.

### Expected Behavior

Genmon updates to new version

### Actual Behavior

Genmon at 1.18.15, Click on Upgrade to Latest Version, download popup appears and it restarts but it's still on 1.18.15 with 1.18.16 still available

### Steps to Reproduce (including precondition)

See above. logs submitted.

### Screenshot or Pictures relating to the problem (if possible)

{Please write here}

### Your Environment

- Generator Model: {Please write here} Generac 20KW
- Genmon Logs: Submit via the About page
- Genmon Version: 1.18.15

jgyates, 2023-02-10:
I received your logs. Nothing jumping out as a major issues. You do have one error that is unusual:

 Error in MyConfig:init: While reading from '/etc/genmon/gengpioin.conf' [line 17]: option 'uselibcallbacks' in section 'gengpioin' already exists : myconfig.py:41

Have you directly edited the file /etc/genmon/gengpioin.conf? It appears that there are two entries for 'uselibcallbacks' in this file. I would remove one of them.

Also, try logging into your pi via ssh and get into the genmon folder:

     cd genmon

then type:

      ./genmonmaint.sh -u
      ./startgenmon.sh restart

This will perform an upgrade and a restart of the software. Post the output of these commands here.

jgyates, 2023-02-15:
I am going to close this issue but if you have any follow up questions or status please post it to this thread.

closed 2023-02-15 by jgyates