jgyates/genmon #858: Pairing Mopeka Pro Issue

bcrazycramer, 2023-02-15:
### Expected Behavior

Paired Mopeka Pro

### Actual Behavior

No actual propane tank reading or incorrect.

### Steps to Reproduce (including precondition)

sudo python3 mopeka_utility.py and adding the sensor bluetooth mac address to the genmopeka.conf file and on the web gui.

### Screenshot or Pictures relating to the problem (if possible)

pi@genmon:~/genmon/OtherApps$ sudo python3 mopeka_utility.py 

NOTE: This program will look for Mopeka Pro Sensors. The SYNC button must be pressed and held for the discovery process to work.

Starting Discovery....
Discovery Stats ReadStats ( Ignored Ad Count: 101, Processed Ad Count: 14, Error Ad Count: 0), Zero Data Ad Count: 0)

Finished Discovery.  Found 1 new sensor(s):

Sensor Address:  CF:0B:34:6A:28:96
Battery Percentage:  100.0%%
Sensor Temperature:  23 C
Tank Level Reading:  0mm

Use the sensor address above as the tank address parameter in the genmon add on settings.

pi@genmon:~/genmon/OtherApps$ 

Using sudo bluetoothctl I don't see any paired devices:

pi@genmon:~/genmon/OtherApps$ sudo bluetoothctl
Agent registered
[CHG] Controller E4:5F:01:DA:25:E4 Pairable: yes
[CHG] Controller 00:1A:7D:DA:71:15 Pairable: yes
[bluetooth]# paired-devices 
[bluetooth]# 

### Your Environment

- Generator Model: Generac
- 
Maintenance :
    Model : 0067292
    Generator Serial Number : 0050045565
    Controller Detected : Evolution, Air Cooled
    Nominal RPM : 3600
    Rated kW : 20 kW
    Nominal Frequency : 60 Hz
    Fuel Type : Propane
    Generator Phase : 1
    Engine Displacement : 999 cc
    Fuel Level Sensor : 0.00 %
    kW Hours in last 30 days : 0.05 kWh
    Fuel Consumption in last 30 days : 0.48 gal
    Total Power Log Fuel Consumption : 1.77 gal
    Run Hours in last 30 days : 0.40 h
    Run Hours in the last year : 1.48 h
    Fuel Level State : Low
- Genmon Logs: Submit via the About page (Submitted 10:37AM PST)
- Genmon Version: 
Version V1.18.16

jgyates, 2023-02-15:
I received your log submission. the submission does contain logs however it does not contain the log for the mopeka add on (/var/log/genmopeka.log).  This is an indicator that the add on is not being loaded when you restart genmon. If you edited the conf files manually rather than via the web UI then I would guess that you did not enable the add on via the genloader file. The web UI will do this for you but you must save your settings on the add on page. Did you edit the conf file manually or did you only use the web UI?

bcrazycramer, 2023-02-15:
I added the sensor tank mac address to the genmopeka.conf file and I also added the mac address in the Mopeka addon in the web interface. Sounds like I just need to add it on the web interface and don't modify the genmopeka.conf file?

jgyates, 2023-02-15:
Correct. Make sure you click save on the specific add on your are using. This will make the web UI enable the add on to be loaded. Just modifying the conf file will not do that.

bcrazycramer, 2023-02-15:
pi@genmon:/var/log$ tail genmopeka.log 
2023-02-15 11:01:37,896 : Error in MyConfig:ReadValue: genmopeka: send_notices: Not a boolean:  : myconfig.py:106
2023-02-15 11:01:38,756 : No valid tank address found: 
pi@genmon:/var/log$

bcrazycramer, 2023-02-15:
[genmopeka]

# number of minutes between polls for tank data.
poll_frequency = 15

# tank address
# This value is MAC address of the sensor. It must be in a format using
# hex numbers separated by colons (i.e. 9a:03:2b:67:25:12).
# multiple tank sensors (up to 4) may be used with commas
# between each address
tank_address = CF:0B:34:6A:28:96

# specifiy the tank type
# Valid types: 20_LB, 30_LB, 40_LB, 100_LB, 200_LB, 120_GAL, 250_GAL,
# 500_GAL, 1000_GAL and Custom
# Custom tank type relies on user defined values of min_mm and max_mm
tank_type = 120_GAL

# optional - integer - Tank min millimeters - minimum millimeter reading of sensor
# used with Custom Tank type
min_mm =

# optional - integer - Tank max millimeters - maximium millimeter reading of sensor
max_mm =

# optional boolean
# this value, if true will send an email notice if the sensor batter is low or if
# no Bluetooth sensors are detected. Outbound email must be setup and working
# with genmon for this feature to work
send_notices = true

[image]

bcrazycramer, 2023-02-15:
I'm not sure why this isn't showing the correct percentage. The tank app is showing 78%

[image]

jgyates, 2023-02-15:
It looks like you have the 120 gallon tank selected. Is this the same tank you have selected in the mopeka app?

jgyates, 2023-02-15:
Here are a few things to try that will allow you to gather some data that may help in determining what is wrong:

You should be able to set the tank level units in the app to centimeters and then use the ./OtherApps/mopeka_utility.py program to display the output from the sensor. This value should be in millimeters so divide the output of the mopeka_utility by 10 and it should be the same as the centimeter output as the app (or close since this value can change between readings).

You can edit the file /etc/genmon/genmopeka.conf and add the following line:

debug = True

save the file and restart genmon. This will put additional data in the /var/log/genmopeka.log file that will be sent when you submit your logs via the about page. Note that you should not run the software all of the time with the debug = True setting as this will reduce the life of your SD card over time.

bcrazycramer, 2023-02-15:
Is there somewhere that we need to specify that this is a horizonal tank? 

2023-02-15 13:53:28,743 : Tank Address = C4:45:32:B9:C6:C1
2023-02-15 13:53:28,747 : min: 38.1 , max: None
2023-02-15 13:53:28,748 : Tank Type: 120_GAL
2023-02-15 13:53:28,748 : Tank Volume Calculated: 147.45
2023-02-15 13:53:28,748 : <Vertical tank, V=34061.996808 m^3, D=30.000000 m, L=42.000000 m, torispherical heads, a=5.080128 m.>
2023-02-15 13:53:29,750 : starting scan
2023-02-15 13:53:45,043 : stopped scan
2023-02-15 13:53:45,043 : Tank Level in mm: 443
2023-02-15 13:53:45,044 : Tank Level in inches: 17.44
2023-02-15 13:53:45,044 : Gallons Left: 47.29
2023-02-15 13:53:45,044 : Tank Percentage: 32.07
2023-02-15 13:53:45,044 : Tank1 = 32.07
2023-02-15 13:53:45,044 : Tank Data = {"Percentage": 32.07, "Tank Name": "Mopeka Sensor Tank"}
2023-02-15 13:53:45,065 : OK

bcrazycramer, 2023-02-15:
This is for the Genmon Mopeka Sensor I was working on this morning. I currently have 2 of these that I'm working on. 

2023-02-15 14:02:24,360 : Tank Address = CF:0B:34:6A:28:96
2023-02-15 14:02:24,364 : min: 38.1 , max: None
2023-02-15 14:02:24,364 : Tank Type: 120_GAL
2023-02-15 14:02:24,364 : Tank Volume Calculated: 147.45
2023-02-15 14:02:24,365 : <Vertical tank, V=34061.996808 m^3, D=30.000000 m, L=42.000000 m, torispherical heads, a=5.080128 m.>
2023-02-15 14:02:25,366 : starting scan
2023-02-15 14:02:40,616 : stopped scan
2023-02-15 14:02:40,617 : Tank Level in mm: 439
2023-02-15 14:02:40,617 : Tank Level in inches: 17.28
2023-02-15 14:02:40,617 : Gallons Left: 46.8
2023-02-15 14:02:40,618 : Tank Percentage: 31.74
2023-02-15 14:02:40,618 : Tank1 = 31.74
2023-02-15 14:02:40,618 : Tank Data = {"Percentage": 31.74, "Tank Name": "Mopeka Sensor Tank"}
2023-02-15 14:02:40,643 : OK

jgyates, 2023-02-15:
Currently the code assumes as 120 gallon tank is vertical. I assume you have a 120 horizontal, correct? if so I can add that in the code.

bcrazycramer, 2023-02-15:
Correct. Most of our tanks are horizontal.

jgyates, 2023-02-15:
I checked in an update that adds and option for horizontal 120 gallon tanks in the add on settings UI. Let me know how that works.

bcrazycramer, 2023-02-15:
Nice!! Now they are both showing about 78 percent.

jgyates, 2023-02-15:
Great, I am glad it is working now. Thanks for helping resolve this. I am going to close this issue now but if you have any additional questions or comments on this topic feel free to post them to this thread.

closed 2023-02-15 by jgyates

bcrazycramer, 2023-02-17:
Can you add a 250 gallon horizontal tank as well?

jgyates, 2023-02-17:
the existing 250, 500 and 1000 are all horizontal orientation. I initially setup the software where all tanks were vertical except 250, 500 and 1000. Now the 120 gal is the only one with both vertical and horizontal orientation.