jgyates/genmon #859: question about notification when unit is charging

grsthegreat, 2023-02-15:
Use the template below if you have an issue or want to report a bug. If you have a question or a feature request you can ignore the questions below. This is a template so feel free to remove any of this text. Try to include the info below when applicable.

***NOTE:*** If you are having issues with your serial connection, please read this page before posting:

I have a general question about seeing notification under generator when my evolution 1 controller is either charging or not charging.

is there a way to add this notification besides state of battery charge?

also, is there a way to set an alert if battery charge gets below a certain level, or does that have to correspond to an alert on the controller

jgyates, 2023-02-15:
The controller has built in alarms that handle low battery voltage. This link describes the charger and give some details on how this alarm works:

https://github.com/jgyates/genmon/blob/3522fb1a6db232cd49305c4396b361b3440cd2e1/genmonlib/generac_evolution.py#L4237

If you are asking if you can get a notification when the battery is charging, no, I don't have that type of notification at the moment as this could give a lot of notices and I am not sure how useful that would be.

grsthegreat, 2023-02-15:
Reason i asked, my battery is usually showing 14.2-14.4 v. When i go out to check it, after scrolling thru dealer menu, it shows charging. Dont know if its due to severe cold weather, or if its continuing charging. It would be nice to look at my generator genmon screen and see if its charging. Just a line under status.

jgyates, 2023-02-15:
Genmon shows the battery charing status of controllers that support this info over modbus. Presently the Evolution Liquid Cooled controller is the only one.

closed 2023-02-15 by jgyates

skipfire, 2023-02-15:
My Evo2 air cooled shows charging status under the unsupported sensors, not sure if that is the right value or not, but I've been utilizing it.

jgyates, 2023-02-15:
Yes. That value is output under unsupported sensors for Evo2.

The Evo Liquid cooled has a bit that you can read that tells the status of the charger.

Evo2 air cooled has a register that I *think* is the CT for the battery charger. If this value is greater than 0.9 then it displays that the battery is charging. This has not been confirmed as accurate as I do not have access to a Evo2 AC that I can measure the charger output, so I put it under unsupported sensors.

skipfire, 2023-02-15:
I'm willing to try to verify if you've got a procedure for me to follow.  From my observation of generator state and battery voltage the `Battery Status (Sensor)` value appears correct, though I'm not sure how it all lines up with the `Battery Charger Sensor`.  Here are some observations I have for the value.

1. After a failed start (RPM sense loss) it went to the high 30's and still said charging.
2. After a successful start it went to the 50's and immediately went to not-charging and it stays not charging for the entire time it is running.
3. After stopping the generator it went briefly to Charging in the 20s range and then dropped to 0.00 and Not Charging
4. During the few minutes before trying to start, I saw values as low as 0.87 and as high as 1.62 all showing charging.

jgyates, 2023-02-15:
Here is how it is displayed now.

On the unsupported sensors section:
Battery Charger Sensor =  the register 05ee divided by 100 and displayed as a floating point value
Battery Status (Sensor) = if the engine is not stopped and not off then display "Not Charging", if the Battery Charger Sensor value is above 0.9 it is charging. This would be when it is running in auto or manual, exercising, cooling down or cranking.

All I know about the charging cycles is in this comment in the code that came from the maintenance manual:

https://github.com/jgyates/genmon/blob/3522fb1a6db232cd49305c4396b361b3440cd2e1/genmonlib/generac_evolution.py#L4237