jgyates/genmon #860: using a hyperv to connect to generac rather than Pi

p15649, 2023-02-18:
Hi; is it possible to run linux on hyperv and connect to the genrator ip address and run the program without setting up a pi?
thnx

skipfire, 2023-02-18:
You have to have something in the generator.  There is a solution to do it with an ESP32 that has a wifi connection and relays the serial data over the network, then GenMon can read from that.  A few users have done it and there are I think 2 different projects that have described the physical setup.  On the ESP32 you run ESPHome with serial relay, if you have Home Assistant it is a really small software setup & configuration effort.

I'm going to be offering hardware setup that can do it in the next 3 to 6 weeks (I'm behind the PintSize.Me hardware).

p15649, 2023-02-18:
If there is a easy hardware solution so I can just connect and add it to
home assistant without doing a lot of effort and coding, that would be
awesome. Just don't have the time to spend a lot of time doing it. Figured
if I can get Hyper-V install Linux and have it just communicate over the
Wi-Fi and get the data, that would have been great as well. Thanks

skipfire, 2023-02-18:
I've not seen anyway to tap into the data available through the MobileLink WiFi module.

As far as I know my PintSize.Me hardware is the only option that doesn't require you to piece together cables and external components yourself.  In March I should have 2 options that require minimal install effort and no custom cabling or anything.  The HAT on a RPi 3A+ (I managed to get 100 of them last week) with preloaded micro-SD card, or the ESP32 unit that can be used as a serial relay.

If you want to look at the existing project for setting up an ESP32 for serial bridge, check out https://github.com/gregmac/Genmon-ESP32-Serial-Bridge.

You can send me a message from https://pintsize.me to be added to the contact list for the ESP32 or preloaded Pi.

jgyates, 2023-02-18:
You can run genmon in any linux environment, with the following caveats: 

1) If you connect your linux system directly to your generator with a serial cable you need to setup the serial port on the linux platform and tell genmon the name of the serial port (i.e. /dev/serial0). The project has utilites to do this on the raspberry pi platform.
2) Don't use the serial port and run genmon in a linux environment but communicate with the generator controller via serial over TCP/IP. This requires the generator to be connected to a device that will conver the serial connection of your generator controller to a network connection (usually Wifi), then genmon will communicate over the network to the serial to netowrk converter. This [wiki article](https://github.com/jgyates/genmon/wiki/Appendix-F----Serial-over-IP) describes the process in more detail. @skipfire is offering an ESP32 based serial bridge solution like the one mentioned in the wiki but is designed to be more turnkey.

jgyates, 2023-02-19:
I am going to close this issue but if you have any follow up question feel free to post them to this thread.

closed 2023-02-19 by jgyates