jgyates/genmon #861: Feature Request: Fuel consumption graph

jmcjclark, 2023-02-21:
I have a suggestion for a future release. A graph displaying fuel consumption over time, similar to the kW output graph.

rwskinner, 2023-02-21:
Not a bad idea, but unless you're running on generator a lot, I don't 
see much use for it, especially if you have a tank monitor that tells 
you true tank percentage.
Genmon only uses an average percentage based on the consumption 
guidelines.  So $50 for a true accurate monitor is cheaper than 
development time and way more useful.

Hopefully my answer isn't too blunt, it's not meant that way, but just 
suggesting a more logical solution, that's all.

Richard

jmcjclark, 2023-02-21:
Thank you Richard. I should have provided more information on my environment. I am using the DIY Fuel Tank Gauge Sensor add on , so there is already a gauge reflecting true tank percentage. The reason for my request is I have multiple appliances on propane including the generator, and feel there would be many with a similar setup. If you feel this isn’t worthwhile for development I totally respect it, as coding is out of my wheelhouse at this point.

jgyates, 2023-02-21:
@jmcjclark,

Thanks for the suggestion. Genmon currently shows the estimated fuel usage in a few ways. On the maintenance page it shows the fuel used in the last 30 days and fuel used in the entire power log. These are calculated at the time they are displayed and only use the power log and the generator model (or user entered fuel rate) for the calculation. The outage page shows the fuel used for a specific outage. If the fuel use were graphed it would show a graph that is very similar to the power graph as most of the fuel rate for different generator models are very close to linear. 

I will think on this but at the moment I am not sure if the value of this feature is worth the effort as this would require quite a bit of new code.

closed 2023-02-21 by jgyates